carla-simulator/carla #7606: I would like to ask: Why does the custom car I added slide to the side when it is stationary?

eggman-1024, 2024-05-07:


GoodarzMehr, 2024-05-07:
Is it on an incline?

MarcelPiNacy-CVC, 2024-05-10:
Hi,

could you provide more info?

Marcel

eggman-1024, 2024-05-10:
> Hi,
> 
> could you provide more info?
> 
> Marcel

okey, I spawn an custom vehicle, the initial yaw I haved set is 0.0, but it`s yaw is changing when I don`t set apply_control to the vehicle.look at this vedio I haved recorded.

https://github.com/carla-simulator/carla/assets/79270109/0dea48bc-3b41-45f7-bf4b-cba4abb48376

eggman-1024, 2024-05-10:
> Hi,
> 
> could you provide more info?
> 
> Marcel
can u see the minor changes...

MarcelPiNacy-CVC, 2024-05-10:
It appears there is some translation too.
This could be a problem with the geometry of the vehicle. Have you checked whether the collision mesh of your vehicle is similar to the defaults in CARLA?

eggman-1024, 2024-05-10:
> It appears there is some translation too. This could be a problem with the geometry of the vehicle. Have you checked whether the collision mesh of your vehicle is similar to the defaults in CARLA?

How should I check? Is it convenient to provide a detailed step? Thank you very much!

MarcelPiNacy-CVC, 2024-05-10:
Opening the mesh asset in the editor should expose the necessary options. Here's a video for reference, the first 30s show how to display the collision mesh: [Link](https://www.youtube.com/watch?v=pkLosy4gdAw)

eggman-1024, 2024-05-10:
> Opening the mesh asset in the editor should expose the necessary options. Here's a video for reference, the first 30s show how to display the collision mesh: [Link](https://www.youtube.com/watch?v=pkLosy4gdAw)

Thank you! But where should I click to view it?
[image]

eggman-1024, 2024-05-11:
[image]

> Opening the mesh asset in the editor should expose the necessary options. Here's a video for reference, the first 30s show how to display the collision mesh: [Link](https://www.youtube.com/watch?v=pkLosy4gdAw)

Hi, I checked the collision mesh of my custom vehicle. It looks similar to the defaults in Carla.

gsmario, 2024-05-15:
Hello, have you followed all the steps from this guide?
https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/#add-a-4-wheeled-vehicle

It could also help us if you record a new video with a different angle/materials in order to see if the vehicle is actually working or not, since your first video is unclear.

And can you post your screenshots like this? so we can see more info.
Thanks in advance.
[image]

eggman-1024, 2024-05-18:
> And can you post your screenshots like this? so we can see more info.
> Thanks in advance.

OK, this is the vedio about the state of my custom car in carla, and the screenshot!Thank you!!!

https://github.com/carla-simulator/carla/assets/79270109/6d917da4-3905-4cb6-ba3b-c65c20516ebd

[image]

eggman-1024, 2024-05-18:
> > And can you post your screenshots like this? so we can see more info.
> > Thanks in advance.
> 
> OK, this is the vedio about the state of my custom car in carla, and the screenshot!Thank you!!!
> 
>  custom_vehicle_vibrate.mp4 
> ![screenshot_for_github](https://private-user-images.githubusercontent.com/79270109/331774481-d7afaa75-0183-44d4-92a7-0cfcd827075f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYwMTc0MjksIm5iZiI6MTcxNjAxNzEyOSwicGF0aCI6Ii83OTI3MDEwOS8zMzE3NzQ0ODEtZDdhZmFhNzUtMDE4My00NGQ0LTkyYTctMGNmY2Q4MjcwNzVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE4VDA3MjUyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzOWViNzM2MGY0YTZiMzVlMDc4MGYyMWM3OGYxYjZkMzc3NzAxMzkyODdhYTgyNWFjZjE1NmI1ZjE2OGU5ZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CqROSSTXGnMunW-q-fG2UzM6yVsrU9E9iO3pVTHYOv8)

In my opinion, it looks like a tiny part of the car got stuck into the ground...but I am not sure....

gsmario, 2024-05-24:
Hello
I can see that you're not following the documentation correctly, since in the documentation says that you need to put linear damping to 0, but you have it at 0.1.
For this, I can't pinpoint your problem, because if you deviate from documentation, anything can fail.
Fel free to correct the problems, following the documentation, and let me know if something fails then.

Thanks for your time.

closed 2024-06-07 by gsmario