carla-simulator/carla #7612: Cannot Change Motorcycle/Bicycle Driver

GoodarzMehr, 2024-05-07:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: I am using the following piece of code to change the driver of motorcycles/bicycles:
```Python
if v_blueprint.has_attribute('driver_id'):
    v_blueprint.set_attribute(
        'driver_id',
        random.choice(v_blueprint.get_attribute('driver_id').recommended_values)
    )
```
However, I have noticed that in practice nothing changes and each motorcycle/bicycle is spawned with the same driver. Could you please confirm if this is the case or if I am doing something wrong?

Blyron, 2024-05-10:
Yeah. We haven't implemented that yet. We will need to do that in Editor. For now at least

00bix, 2024-05-10:
Out of curiosity: When is spawn this bike: https://carla.readthedocs.io/en/0.9.15/catalogue_vehicles/#diamondback-century
I get this rider : 
[image]

... How come the vehicles in the catalgoue are not the vehicles i can spawn? Am I missing something?

DnDare, 2024-05-10:
It's the same vehicle, but we updated the rider.

GoodarzMehr, 2024-05-11:
Thanks! I saw it in multiple example scripts so I was under the impression that it would change the rider.

I'll leave this issue open for now. Please let me know if/when it gets implemented, since it would help diversify scenarios a tiny bit more.

closed 2024-05-17 by Madecu