carla-simulator/carla #7621: legs in walker

Montyro, 2024-05-10:
Hi, first of all, thank you for sharing your work with the world. 

I have been playing around with walker bones and believe they have some issues. I have been trying to set up their rotations but many joints seem to be oriented in a weird way from their parent as if they were placed in the model with some offset from the "0,0,0" relative rotation. Maybe I'm understanding something wrong or trying to perform rotations in a non-intended way, if so, sorry for opening the issue.

Here's the issue:

CARLA version: 9.15
Platform/OS: Windows 11
Problem you have experienced: Walkers have inconsistent relative rotations.
What you expected to happen: If I set all rotations to 0 for walkers, while keeping locations, the person should be symmetrical, and each bone to preserve the rotation of their parents, but it doesn't, it loses symmetry (different behaviours on each crl_arm, crl_thigh, crl_forearm, crl_legs, and crl_foot)

[image]

Steps to reproduce:
Spawn a pedestrian, and set all rotations to 0 while keeping the same relative locations. This is my bit of code:

    all_bones = ped.get_bones() 
    new_bones = []

    for bone in all_bones.bone_transforms:
        if bone.name == 'crl_root':
            rotation = carla.Rotation(0,0,90)
        else:
            rotation = carla.Rotation(0,0,0)
         
        location = [bone.relative.location.x, bone.relative.location.y, bone.relative.location.z ]

        new_bones.append((bone.name,carla.Transform(rotation=rotation,location=carla.Location(x=location[0],y=location[1],z=location[2]))))

    ped.set_bones(carla.WalkerBoneControlIn(new_bones))

DnDare, 2024-05-10:
Hi, 
When we import a pedestrian to Unreal, we make a rigging from Maya with a binding pose with all bones reset to zero. T-pose means all rotations at zero.
But Unreal changes the skeleton and the rotations at the skeletal definitions, we have no way to control this. 

If you want to know which values Unreal has assigned, you'll have to review them at the project.

look for the SK_Pedestrian_Generan   skeletal definition, you'll find the T-pose rotation values:
[image]

Maybe you can find this values from the package, but I'm not 100% sure

Montyro, 2024-05-10:
Thanks a lot for the quick and detailed answer. Will check in the unreal project, hopefully I can work around this.

Montyro, 2024-05-11:
Hi again, I have been playing with these unreal values, and by applying the inverse relative rotation you can get it to the right 0 position, so that's good, at least for legs which are the bones I used to test it. I'm now closing the issue.

Thanks a lot!

closed 2024-05-11 by Montyro

zybermonk, 2024-12-13:
Hi @Montyro, 

I was playing with the `WalkerBoneControlIn()` as well, and your approach is similar to mine w.r.t using the inverse relative transforms to get true zero pose.

However, I noticed it is only working for T pose (setting all rotations 0) and individual bones BUT not for a desired new pose. Upon setting a new full body pose, the pose is crooked in CARLA.

I am curious to know how you handled the inverse_matrix because
-  when all rotations are 0 in unreal, the pose is not T pose (without using the inverse method) and its like seen in your picture, and at this pose, all local frames are identical. 
- when we set a T pose and check the local frames, they are not all identical, especially the limbs have random local frames.

Would like to know your thoughts and if you are using `get_inverse_matrix()` function from CARLA or using Unreal to construct your own inverse matrix, and if you were able to make a full body pose by any chance.

Thank you

cucdengjunli, 2025-04-13:
bone_control.bone_transforms = bone_transforms
                
                # 应用骨骼控制
                walker.set_bones(bone_control)
                walker.show_pose()