carla-simulator/carla #7623: LINK: fatal error LNK1181: carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\sqlite3.lib'  cannot open input file.

wkdwnsgur070, 2024-05-10:
OS: Window 11
CMake: 3.81
CMake generator: Visual Studio 16 2019

  [40/46] UE4Editor-Carla.lib
     C:\Users\wkdwn\carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla.lib 라이브러리 및 C:\Users\wkdwn\carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\Carla\UE4Editor-Carla.exp 개체를 생성하고 있습니다.
  [41/46] UE4Editor-StreetMapRuntime.dll
  LINK : fatal error LNK1181: 'C:\Users\wkdwn\carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\sqlite3.lib' cannot open input file..
  [42/46] UE4Editor-CarlaUE4.dll
  LINK : fatal error LNK1181: 'C:\Users\wkdwn\carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\sqlite3.lib' cannot open input file..
  [43/46] UE4Editor-Carla.dll
  LINK : fatal error LNK1181: 'C:\Users\wkdwn\carla_simulator\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\sqlite3.lib'  cannot open input file.
make: *** [CarlaUE4Editor] ERROR 6

I got this error message during my installation.
How can i fix this problem?

Lucky-Lucus, 2024-05-12:
I had the same issue when executing `make launch` (with `GENERATOR="Visual Studio 17 2022` since I am using VS 2022) on Windows 11 and with Carla 0.9.15

I managed to fix it by locally building sqlite3.lib file using [this](https://gist.github.com/zeljic/d8b542788b225b1bcb5fce169ee28c55) method (does work for Windows 11 as well) and then pasting the resulting sqlite3.lib file in the directory where it is missing (`carla_root_folder\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\`)

wkdwnsgur070, 2024-05-13:
Thank you for your help!

MarcelPiNacy-CVC, 2024-05-15:
Hi,

Could you provide the entire build log?

Marcel