carla-simulator/carla #7625: The traffic light remains red for longer than the time observed with get_red_time()

seominseoktech, 2024-05-11:
- CARLA version: 0.9.15
- Platform/OS: Windows

Currently in Town10, I am using `world.get_snapshot().timestamp.elapsed_seconds` to monitor whether the traffic lights stay for the designated duration.

Upon investigation, I found that the yellow and green lights operate exactly for the set duration, but the red light remains unchanged and persists beyond the designated time (default 2 seconds).

Below is the code I used, and upon checking the saved CSV file, it appears that even though 2 seconds have elapsed, the traffic light remains set to red.

[image]

```python
from typing import Dict, List
import pandas as pd
import time
import carla

def traffic_light_to_dict(traffic_light: carla.TrafficLight) -> Dict: 
    time = world.get_snapshot().timestamp.elapsed_seconds
    id = traffic_light.id
    location = traffic_light.get_location()
    state = str(traffic_light.get_state())

    d = {
        "time": time,
        "id": id,
        "state": state,
        "red_time": traffic_light.get_red_time(),
        "yellow_time": traffic_light.get_yellow_time(),
        "green_time": traffic_light.get_green_time(),
        "elapsed_time": traffic_light.get_elapsed_time(),
        "x": location.x,
        "y": location.y,
        "z": location.z
    }
    
    return d

def dict_to_dataframe(d, features: List) -> pd.DataFrame:
    df = pd.DataFrame([d])
    return df[features]

list_actor = world.get_actors()

traffic_light_data = []

start_time = time.time()

while(time.time() - start_time < 60):
    for actor_ in list_actor:
        if isinstance(actor_, carla.TrafficLight):
            if actor_.id == 12:
                traffic_light_data.append(dict_to_dataframe(traffic_light_to_dict(actor_), features=["time", "id", "state", "elapsed_time"]))
 
traffic_light_df = pd.concat(traffic_light_data)
traffic_light_df.to_csv('traffic_light_data.csv', index=False)
```

seominseokt, 2024-05-22:
- [Traffic lights stay red for long time #6932](https://github.com/carla-simulator/carla/issues/6932)
- [Control Red light doesn't work #6121](https://github.com/carla-simulator/carla/issues/6121)

Traffic lights operate in groups, and when one group's cycle ends, it waits at a red light until the other group's cycle finishes, resulting in a longer duration of the red light.

For example, if there are two groups (Group 1 and Group 2) and each maintains green for 10 seconds, yellow for 3 seconds, and red for 2 seconds:

When Group 1's cycle ends, Group 2's cycle begins, during which Group 1 stays on red.

Thus, Group 1's red light lasts for 2 seconds (Group 1's red) + 10 seconds (Group 2's green) + 3 seconds (Group 2's yellow) + 2 seconds (Group 1's red) = 17 seconds.

closed 2024-05-22 by seominseokt

lingyixia, 2024-09-25:
> * [Traffic lights stay red for long time #6932](https://github.com/carla-simulator/carla/issues/6932)
> * [Control Red light doesn't work #6121](https://github.com/carla-simulator/carla/issues/6121) [...]

So do you know how to ignore the influence of group on traffic light？There are only on vehicle in my scenario so I do not need to consider traffic accidents when ignoring group。

lingyixia, 2024-09-25:
> * [Traffic lights stay red for long time #6932](https://github.com/carla-simulator/carla/issues/6932)
> * [Control Red light doesn't work #6121](https://github.com/carla-simulator/carla/issues/6121) [...]

So do you know how to ignore the influence of group on traffic light？There are only on vehicle in my scenario so I do not need to consider traffic accidents when ignoring group，but I need a  time-shorter red lights。

seominseokt, 2024-10-05:
@lingyixia  I’m sorry. I just need to calculate the operation time for each color of the traffic light, and I don’t know how to ignore the influence.