carla-simulator/carla #7630: light need to be rebuilt in night in 0.9.15

zhoujiawei3, 2024-05-11:


zhoujiawei3, 2024-05-11:
CARLA version: 0.9.15
Platform/OS:windows11
Problem you have experienced: when time change to night, not all the light is on. The UE4 editor appears"light need to be rebuilt",and light behavior is different from that in 0.9.14
0.9.14:
[image]

0.9.15:
[image]

Blyron, 2024-05-12:
Use this to fix:
https://www.youtube.com/watch?v=iPN2hWzfmP8&ab_channel=qubodupDev
This does not happen if you makes a package due to Lightning is built during the packaging process

closed 2024-05-12 by Blyron

zhoujiawei3, 2024-05-12:
when I follow the instruction in the video, it appears [image] And the problem still happens.