carla-simulator/carla #7633: Boost is downloaded twice during build

Daraan, 2024-05-13:
CARLA version: latest dev (9037124f594daa09f745f00d7125a513cac8aaaf) 
Platform/OS: Ubuntu 22.04
Python: 3.10 & 3.7

**Problem you have experienced:**

When making a new build it currently downloads boost twice and creates two tar files.
I assume this is not a wanted behavior.

---

Output:

```sh
Util/BuildTools/Linux.mk:163: warning: overriding recipe for target 'downloadplugins'
Util/BuildTools/Linux.mk:142: warning: ignoring old recipe for target 'downloadplugins'
Setup.sh: Retrieving boost.
--2024-05-13 10:52:08--  https://archives.boost.io/release/1.80.0/source/boost_1_80_0.tar.gz
Resolving archives.boost.io (archives.boost.io)... 146.75.119.52
Connecting to archives.boost.io (archives.boost.io)|146.75.119.52|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136670223 (130M) [application/octet-stream]
Saving to: ‘boost_1_80_0.tar.gz.8’

boost_1_80_0.tar.gz.8        100%[=============================================>] 130,34M  68,4MB/s    in 1,9s    

2024-05-13 10:52:10 (68,4 MB/s) - ‘boost_1_80_0.tar.gz.8’ saved [136670223/136670223]

Elapsed Time downloading from boost webpage: 2 seconds
Setup.sh: Using boost backup
--2024-05-13 10:52:11--  https://carla-releases.s3.us-east-005.backblazeb2.com/Backup/boost_1_80_0.tar.gz
Resolving carla-releases.s3.us-east-005.backblazeb2.com (carla-releases.s3.us-east-005.backblazeb2.com)... 149.137.137.254
Connecting to carla-releases.s3.us-east-005.backblazeb2.com (carla-releases.s3.us-east-005.backblazeb2.com)|149.137.137.254|:443... connected.
HTTP request sent, awaiting response... 200 
Length: 136670223 (130M) [application/x-gzip]
Saving to: ‘boost_1_80_0.tar.gz.9’

boost_1_80_0.tar.gz.9        100%[=============================================>] 130,34M  28,2MB/s    in 5,4s    

2024-05-13 10:52:18 (23,9 MB/s) - ‘boost_1_80_0.tar.gz.9’ saved [136670223/136670223]

Elapsed Time downloading from boost carla backup in backblaze: 7 seconds
```

MarcelPiNacy-CVC, 2024-05-14:
Hi,

Could you share a screenshot of your Build folder?

Marcel

Daraan, 2024-05-14:
yes sure

[image]

MarcelPiNacy-CVC, 2024-05-14:
Right, I think I see the issue.

At line 97 in <CARLA>/Util/BuildTools/Setup.sh you will see `wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" -O || true`  
Replace it with this:  
`wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" -O ${BOOST_PACKAGE_BASENAME}.tar.gz || true`

Daraan, 2024-05-14:
Thank you for the reply, makes sense to force the name

1. however from my experiment just now `-O` still appends `.#` duplicates.
2. I also it with `-nc` to not duplicate it and compared `sha256sum` on the tar file and its the same as written in the global variable, however the if-clause in 102 is still entered, resulting in a second download. My shell script knowledge is to weak currently for a fast debugging.

PR #7643 (2024-05-15): Fix double dowload of boost backup due to wrong string comparision with sha256sum

#### Description

Fixes  #7633

The `sha256sum` check on the boost file returns a string in the format `<hash> <filename>` in the build script the string comparison fails because it only is checked against <hash>.

Applied solution and problem description see: https://unix.stackexchange.com/questions/265740/suppress-filename-from-output-of-sha512sum

Also prevents duplication of multiple boost tar files due to automatic renaming with numbers Credit to @MarcelPiNacy-CVC

---

#### Where has this been tested?

  * [... cut]
closed 2024-05-15 by Blyron