carla-simulator/carla #7636: Problem when trying to attach a spawned camera to a vehicle

Carmineh, 2024-05-13:
**CARLA version**: 0.9.15
**OS**: Windows

When trying to spawn a camera and attach it to a car it gives an error and code doesn't run.

**Error**
>Failed to spawn sensor: rpc::rpc_error during call in function spawn_actor_with_parent

**Code**: Here's the snippet of the code created (If I spawn the camera without using the attach_to works fine)

```python
import glob
import os
import sys

try:
    sys.path.append(
        glob.glob(
            "../carla/dist/carla-*%d.%d-%s.egg"
            % (
                sys.version_info.major,
                sys.version_info.minor,
                "win-amd64" if os.name == "nt" else "linux-x86_64",
            )
        )[0]
    )
except IndexError:
    pass

import carla
import random
import time

directory_name = "output_data"
current_dir = os.getcwd()
if not os.path.exists(directory_name):
    os.makedirs(directory_name)
    print(f"Directory '{directory_name}' created in '{current_dir}'")
else:
    print(f"Directory '{directory_name}' already exists in '{current_dir}'")

IMG_FOLDER = os.path.join(current_dir, directory_name, "/")
IMG_WIDTH = 480
IMG_HEIGHT = 480

TOTAL_PHOTOS = 0

def save_image(image):
    image.save_to_disk(IMG_FOLDER + "%08d" % image.frame)
    TOTAL_PHOTOS += 1
    print(f"Total images saved: {TOTAL_PHOTOS}")

actor_list = []
try:
    client = carla.Client("127.0.0.1", 2000)
    client.set_timeout(2.0)

    world = client.get_world()

    blueprint_library = world.get_blueprint_library()
    
    vehicle_blueprint = blueprint_library.find('vehicle.tesla.model3')  # Scelta del modello dell'auto
    vehicle_spawn_point = random.choice(world.get_map().get_spawn_points())
    vehicle = world.spawn_actor(vehicle_blueprint, vehicle_spawn_point)
    actor_list.append(vehicle)

    sensor_blueprint = blueprint_library.find("sensor.camera.rgb")
    sensor_blueprint.set_attribute("image_size_x", f"{IMG_WIDTH}")
    sensor_blueprint.set_attribute("image_size_y", f"{IMG_HEIGHT}")
    sensor_blueprint.set_attribute("fov", "110")
    sensor_spawn_point = carla.Transform(
        carla.Location(x =0.5 , z = 1.5)
    )

    try:
        time.sleep(2)
        sensor = world.spawn_actor(sensor_blueprint, sensor_spawn_point, attach_to=vehicle)
        actor_list.append(sensor)
    except RuntimeError as e:
        print("Failed to spawn sensor: {}".format(e))
        
    time.sleep(5)
finally:  # Una volta terminato il tempo distrugge gli attori creati
    print("destroying actors")
    for actor in actor_list:
        actor.destroy()
    print("done.")
```

The code basically spawn a car and camera attached to it, the other things are just for a university project I'm doing. Thanks in advance

Blyron, 2024-05-13:
Could you send full log?

Carmineh, 2024-05-13:
```
Directory 'output_data' created in 'C:\Users\rocco\OneDrive\Desktop\DMRC'
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.15 
WARNING: Simulator API version  = 0.9.15-177-g4da350ab2-dirty 
ActorBlueprint(id=vehicle.tesla.model3,tags=[vehicle, tesla, model3])
ActorBlueprint(id=vehicle.lincoln.mkz_2020,tags=[vehicle, lincoln, mkz_2020])
Actor(id=174, type=vehicle.tesla.model3)
Actor(id=175, type=vehicle.lincoln.mkz_2020)
Failed to spawn sensor: rpc::rpc_error during call in function spawn_actor_with_parent
Vehicle Autopilot: OFF
destroying actors
done.
```

I tried to change version but nothing, I haven't found anything so I tought it's normal? 
Just to make some other things clear, that's an unreal engine version of CARLA clean, we only took a town and changes some road sign.

Blyron, 2024-05-13:
I recommend you to do a pip 
# Inside the deliverable folder run:
pip3 uninstall carla
pip3 install -r requirements.txt
and then a make PythonAPI and try again

Carmineh, 2024-05-14:
I did that but when I do make PythonAPI it gives me an error

```
Creazione della libreria build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib e dell'oggetto build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.exp
libcarla.obj : error LNK2001: simbolo esterno png_destroy_write_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_bKGD non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_cHRM_fixed non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_pHYs non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_swap_alpha non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_text non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_invert_mono non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_sRGB non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_mem_level non risolto
libcarla.obj : error LNK2001: simbolo esterno png_get_io_ptr non risolto
libcarla.obj : error LNK2001: simbolo esterno png_create_write_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_gAMA_fixed non risolto
libcarla.obj : error LNK2001: simbolo esterno png_write_info non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_buffer_size non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_tIME non risolto
libcarla.obj : error LNK2001: simbolo esterno png_write_row non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_pCAL non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_PLTE non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_sBIT non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_oFFs non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_filter non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_iCCP non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_hIST non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_level non risolto
libcarla.obj : error LNK2001: simbolo esterno png_destroy_read_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_IHDR non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_strip_alpha non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_strategy non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_write_fn non risolto
libcarla.obj : error LNK2001: simbolo esterno png_create_info_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_method non risolto
libcarla.obj : error LNK2001: simbolo esterno png_write_end non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_window_bits non risolto
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 33 esterni non risolti
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\rocco\carla\PythonAPI\carla\dist"!
```

Blyron, 2024-05-14:
Delete your Build folder and try again

Carmineh, 2024-05-14:
Same error, nothing changed

Carmineh, 2024-05-14:
Yes it's the source file, I tried again the "pip install -r requirements.txt" and it says that no module has been found for shapely, the make clean gives error too...

Blyron, 2024-05-14:
Do the pip install -r requiremets.txt inside PythonAPI example folder

Carmineh, 2024-05-14:
Nothing changed, I send you the full log just to see everything
```
C:\Users\rocco\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Users\rocco\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  22
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\Users\rocco\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\Users\rocco\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
        1 file copiati.
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --zlib-install-dir "C:\Users\rocco\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\Users\rocco\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\Users\rocco\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\Users\rocco\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\Users\rocco\carla\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 22
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\Users\rocco\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\Users\rocco\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 file copiati.
        1 file copiati.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\Users\rocco\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 file copiati.
        1 file copiati.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\Users\rocco\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 file copiati.
        1 file copiati.
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\Users\rocco\carla\Build\"
0 File copiati
0 File copiati
        1 file copiati.
        1 file copiati.

[... 139 lines cut ...]
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_buffer_size non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_tIME non risolto
libcarla.obj : error LNK2001: simbolo esterno png_write_row non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_pCAL non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_PLTE non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_sBIT non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_oFFs non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_filter non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_iCCP non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_hIST non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_level non risolto
libcarla.obj : error LNK2001: simbolo esterno png_destroy_read_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_IHDR non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_strip_alpha non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_strategy non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_write_fn non risolto
libcarla.obj : error LNK2001: simbolo esterno png_create_info_struct non risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_method non risolto
[... cut ...] on risolto
libcarla.obj : error LNK2001: simbolo esterno png_set_compression_window_bits non risolto
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 33 esterni non risolti
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\rocco\carla\PythonAPI\carla\dist"!
```

Blyron, 2024-05-14:
It seems libpng is not being installed

Carmineh, 2024-05-14:
I installed it manually but nothing. Should I try the clean installation?

Blyron, 2024-05-14:
Yeah...

Carmineh, 2024-05-14:
I tried it but nothing, did more thing than before but still no fix, it gives same error. I'll try a clean installation then

duong2000x10, 2024-05-16:
I had the same error, and the reason was from my Carla client library. I'm not sure if that is your reason, but perhaps you can try installing the .whl file in the PythonAPI/carla/dist/

closed 2024-06-19 by Carmineh