carla-simulator/carla #7642: light need to be rebuilt in night in 0.9.15

zhoujiawei3, 2024-05-14:
CARLA version: 0.9.15
Platform/OS:windows11
Problem you have experienced: when time change to night, not all the light is on. The UE4 editor appears"light need to be rebuilt",and light behavior is different from that in 0.9.14
0.9.14:
[image]
0.9.15：
[image]

the method in this https://www.youtube.com/watch?v=iPN2hWzfmP8&ab_channel=qubodupDev doesn't work. The problem still exists. When I choose only to rebuild lighting, this instruction appears 
[image]

zhoujiawei3, 2024-05-14:
[image]

zhoujiawei3, 2024-05-14:
the light need to be rebuilt happens when time changes to night. And I build Carla on another computer, the problem still exists

gsmario, 2024-05-17:
Hello, and thanks for your time. I've fixed the problem you say about the museum emissive textures, it will be on the next release.

About the lights needs to be rebuilt, this message appears every time you move anything static on the scene, but since we're using a dynamic light, this is not necessary.

Thanks again for your help finding this problem!

zhoujiawei3, 2024-05-17:
> Hello, and thanks for your time. I've fixed the problem you say about the museum emissive textures, it will be on the next release.
>
> About the lights needs to be rebuilt, this message appears every [...]

Thanks for your reply, but I think not only museum light is different from 0.9.14. Many lights in building also different. Just see the photo I post, the building behind seems behaves differently. In my opiniom  Maybe 13610 light not open, same as the message light need to be rebuilt. 

And response to your explanation about light needs to be rebuilt problem, I run the same script to generate image. They all have world.unload_map_layer(carla.MapLayer.ParkedVehicles). Maybe this means removing static on the scene? But when I run the script in 0.9.14, the message don't appear. It only happens in 0.9.15

zhoujiawei3, 2024-05-17:
Should I change the light from fix to movable to make all the lights on ? 

> Reference in ne
Should I change the light from fix to movable to make all the lights on ? 
[image]

zhoujiawei3, 2024-05-17:
Is there any way to change the source about emissive textures same from 0.9.14

gsmario, 2024-05-24:
Hello, I'm still looking at this, since moving between versions take some time, if you think that the problem is the material, you can create a simple material with emissive texture and test with it.
I'll update you later, thanks for your time!

gsmario, 2024-06-28:
I have been testing the error and I cannot reproduce it, the lights look fine to me. The source of the emissive textures is the same as version 9.14, so it shouldn't be a problem.
It may be that in the next release your problem will disappear.
Thanks for your time.

closed 2024-06-28 by gsmario