carla-simulator/carla #7646: No CarlaUE4.sh file

Ahnhojin1223, 2024-05-15:
CARLA version: CARLA Server Version: 0.9.15-191-g12f09009a-dirty / CARLA Client Version: 0.9.15
Platform/OS: Ubuntu 20.04

I build carla using https://carla.readthedocs.io/en/latest/build_linux/
When I launch it using make launch command it works.
After I finished build and re-open carla there is no CarlaUE4.sh.
So I can't open carla server using ./CarlaUE4.sh

Am I always open carla server using "make launch" command?
There is no CarlaUE4.sh file in github I clone in https://carla.readthedocs.io/en/latest/build_linux/???

Blyron, 2024-05-15:
CarlaUE4.sh is generated in a package, if you want to use sh you will need to create a package using make package which will be stored in Dist folder, or download one from our releases

Ahnhojin1223, 2024-05-23:
> t

Ahnhojin1223, 2024-05-23:
> CARLA version: CARLA Server Version: 0.9.15-191-g12f09009a-dirty / CARLA Client Version: 0.9.15 Platform/OS: Ubuntu 20.04
>
> I build carla using https://carla.readthedocs.io/en/latest/build_linux/ [...]

Ahnhojin1223, 2024-05-23:
Then can I 

> CarlaUE4.sh is generated in a package, if you want to use sh you will need to create a package using make package which will be stored in Dist folder, or download one from our releases

Then can I merge those two folder???

PabloVD, 2024-05-24:
They are not intended to be merged. Depending on the use you want to make of Carla, you may choose either using a package or building from source.
If you just want to run Carla with the current features and assets, making use of the Python API, just download a package, which does not requires the Unreal editor and it is much simpler to get started.
Instead, if you plan to add new features modifying the source code, you should build from source with the Unreal editor. From that, you can also generate a package.

closed 2024-07-29 by Blyron