carla-simulator/carla #7650: How to replay and record from a driving perspective

ldt17, 2024-05-15:
May I ask how to replay and record from a driving perspective.

I conducted an experiment using version 0.9.14 by setting up cameras to simulate normal driving. However, I found that some of the data was not collected and I want to replay the record again. But using the start.replaying.py file will track the vehicle from a top view perspective, which is not what I want

ldt17, 2024-05-15:
In other words, may I ask if I can collect the data I need again during the replay process

ldt17, 2024-05-16:
@PatrickPromitzer Thank you for your reply！

However, my experiment included other external data such as eye tracker, so I was wondering if it was possible to re collect the data through replay recording instead of conducting a new experiment. 

Thank you !

PatrickPromitzer, 2024-05-16:
If I understood it right, you have 3 types of data.

1. Carla recording
- Actor data
2. Sensor data
- Camera
3. External data
- Eye tracker

The [sensor_synchronization.py](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/sensor_synchronization.py) example it will spawn Actors, add sensors to the vehicle, and collect the sensor data.

The Carla recording is spawning the Actors (if I understood it right), but doesn't add Sensors.
In that case, you need to add the sensors yourself, after starting the replay of the recording.

Python Script
1. start replay 
2. activate synchronization mode (I am not sure if the replay supports that) 
3. add Sensors listener
4. world.tick() (send simulation update request)
5. collect the data
6. save Sensor data and Simulation data
7. repeat 4-6 until the replay finished

The tricky part could be, if you have many similar Actors and can't find the Actor you need. For that reason it is a better to add a role name to the important Actors. (ScenarioRunner does that)

Is this the solution you want, or do you want something else?
It that is not what you want, it would be nice to give an example.

ldt17, 2024-05-16:
@PatrickPromitzer Yes, as you said, I am trying your method. Thank you very much！

closed 2024-05-31 by Madecu