carla-simulator/carla #7652: CARLA C++ client example/easy_import_cmake_include

xavisolesoft, 2024-05-15:


MarcelPiNacy-CVC, 2024-05-15:
I agree, we also should explicitly handle the case of installing with CMake.

xavisolesoft, 2024-05-15:
Add the end we need to include some of the headers included here to CARLA UE5 plugin:
https://github.com/carla-simulator/carla/blob/ue5-dev/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Carla.Build.cs#L128

xavisolesoft, 2024-05-15:
> I agree, we also should explicitly handle the case of installing with CMake.

You mean installing the CARLA C++ client with all the headers to be able to distribute it out of the box of CARLA?
It could make sense for CARLA Release, to prevent the customers to download and build all CARLA to implement a C++ client application.

closed 2024-10-18 by Blyron