carla-simulator/carla #7654: CarlaUnreal repository not found

ShruthiVM, 2024-05-15:
Hello all,

**CARLA version:** 0.9.13

**Platform/OS:** Windows

**Problem you have experienced**: I have been trying to clone the Carla fork unreal engine using the syntax git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git but it keeps throwing the error repository not found. Is there a change in the path to this repository?

**What you expected to happen:** Download successfully after GitHub authentication.

**Steps to reproduce:** The steps are as follows in the command prompt run as an administrator
[image]

Thank you for the help in advance.

Best regards,
Shruthi

-->

seominseoktech, 2024-05-16:
Please use this file. It's a file I received in the Carla Discord channel.

https://drive.google.com/file/d/1zWEoCWDDwuPRZjTK3Q1YY4KA-b3B2iCX/view?usp=sharing

PabloVD, 2024-05-17:
The repository is still the same as before. Have you linked your github account to the Unreal Engine's account?
If not, please follow [this guide](https://www.unrealengine.com/en-US/ue-on-github) for that.

ShruthiVM, 2024-05-22:
Hello Seominseokt

> Please use this file. It's a file I received in the CARLA Discord channel.
> 
> https://drive.google.com/file/d/1zWEoCWDDwuPRZjTK3Q1YY4KA-b3B2iCX/view?usp=sharing

Thank you very much for the link. I'll use it if this issue does not get solved.

ShruthiVM, 2024-05-22:
Hello Pablo,

> The repository is still the same as before. Have you linked your github account to the Unreal Engine's account? If not, please follow [this guide](https://www.unrealengine.com/en-US/ue-on-github) for that.

I had connected my github account to the Unreal engine's account.

Below is the screenshot of the account linking.

[image]

Also, before the download, the authentication is successful from the github to proceed with the download. Screen shot below for the same.

[image]

I have installed Carla many times before and I have never faced this issue. That is why I want to know if I'm missing out anything new or if anything has changed from before. 

Thank you for your response and the help.

ShruthiVM, 2024-05-22:
Hello Pablo,

Could you at least provide a zip file of the cloning folder so I can continue with other steps? The one provided by Seominseokt has some missing files in the build file and throws error in GenerateProjectFiles.bat

ShruthiVM, 2024-05-22:
Hello Pablo,

I resolved the issue. The problem was that I had not joined the epic games organization through the link provided though I had linked the accounts. Now that I did, the download worked perfectly. Thank you for the help.

closed 2024-05-22 by ShruthiVM

SimonSaysGiveMeSmile, 2024-06-23:
> Please use this file. It's a file I received in the CARLA Discord channel.
> 
> https://drive.google.com/file/d/1zWEoCWDDwuPRZjTK3Q1YY4KA-b3B2iCX/view?usp=sharing

Thank you, Pablo! This is really helpful, God bless you!

krikitup, 2024-11-04:
> Hello Pablo,
> 
> I resolved the issue. The problem was that I had not joined the epic games organization through the link provided though I had linked the accounts. Now that I did, the download worked perfectly. Thank you for the help.

Hi, could you share which link you are talking about? I am trying to install Carla, and having similar issue

AbdessamadElK, 2024-11-20:
Hello, for anyone facing this problem, please thoroughly follow [https://www.unrealengine.com/en-US/ue-on-github](this guide) until the last step before trying to clone the repository. You should receive an email from GitHub containing a link to join @EpicGames.