carla-simulator/carla #7656: Using the manual_control.py script, pygame displays a black screen

Alexued, 2024-05-16:


Alexued, 2024-05-16:
When I click play in UE4editor to run Carla, the pygame window that pops up when running python3.8 manual_control.py in another terminal shows a black screen. Here is the picture:
[image]

Alexued, 2024-05-16:
[image]

PabloVD, 2024-05-17:
Could you bring us more information on your issue? Which version of CARLA are you using? Are you in WIndows or Linux? Did you build CARLA following [the documentation](https://carla.readthedocs.io/en/latest/build_linux/)?

Alexued, 2024-05-20:
I am using version 0.9.15 of carla, running on Ubuntu18.04, and build from source

PabloVD, 2024-05-24:
Could you share the output of the shell where the manual control is running? Are there any errors?
Does the ego vehicle correctly spawn? You can check that for instance by inspecting the scene with the spectator.

vijaygovindappa, 2025-02-10:
facing the same issuse as @Alexued . I am running docker compose up to run carla server 0.9.15 . And i am using galactic ros bride to get the ros2 messages. But after i launch pygame window is blank. 
The only error i am getting is regarding sound card being not found,

[image]

vijaygovindappa, 2025-02-10:
[image]