carla-simulator/carla #7666: Fix Pedestrians colision Gen3

Madecu, 2024-05-16:
Gen3 pedestrians have no  DeathAnim yet, first we must modify the BP_walker, so that it can pick different animation sets depending on it's generation

[image]

DnDare, 2024-05-19:
We'd try to retarget GEN2 DeathAnimations to GEN3 skeletal definition,  UE5 retargeting tools have evolved since UE4. 
*Whisfull thinking,  Maybe, just retargeting would be enough, without modifying BP_Walker. We have to research

DnDare, 2024-09-18:
[image]
[image]
[image]

closed 2026-11-12 by Madecu