carla-simulator/carla #7675: Carla crash with quality mode low and RenderOffScreen

Ahmed-Radwan094, 2024-05-17:
CARLA version: 0.9.15
Platform/OS: linux 20.04, NVIDIA GeForce RTX 3080, Driver Version: 535.171.04   CUDA Version: 12.2
The problem you have experienced:
-  Carla repeatedly crashes when running in low-quality mode and rendering off-screen (./CarlaUE4.sh -quality-level=Low -RenderOffScreen). 
- I reproduced the same issue with different machines (GPUs and Cuda versions) running the same OS. 
- On the other hand, Carla doesn't crash with ( ./CarlaUE4.sh -quality-level=Low, ./CarlaUE4.sh, ./CarlaUE4.sh -RenderOffScreen) while running the exact same code.

Can someone direct me to what could be the issue as running Carla in low quality with no screen would help me train RL agents more efficiently?

Thank you.

Blyron, 2024-05-17:
Which maps are you trying?

Ahmed-Radwan094, 2024-05-17:
Town10, towno4

Ahmed-Radwan094, 2024-05-17:
Yes, as I remove some few layers from the map.

Blyron, 2024-05-17:
NVIDIA Drivers?
Could you share the script you are using to use maps? are you using sensor.imagetodisk function?

Ahmed-Radwan094, 2024-05-17:
NVIDIA Driver Version: 535.171.04

Ahmed-Radwan094, 2024-05-17:
`# save world and map from Carla
        self.world = self.client.load_world(self.config['carla']['desired_map'])
        # remove the prop from map to avoid occlusions in bev images
        self.world.unload_map_layer(carla.MapLayer.Props)
        self.world.unload_map_layer(carla.MapLayer.Foliage)
        # remove the traffic lights and poles from map to avoid occlusions in bev images
        traffic_lights_in_the_map = self.world.get_environment_objects(carla.CityObjectLabel.TrafficLight)
        traffic_lights_in_the_map_ids = {traffic_light.id for traffic_light in traffic_lights_in_the_map}
        self.world.enable_environment_objects(traffic_lights_in_the_map_ids, False)
        poles_in_the_map = self.world.get_environment_objects(carla.CityObjectLabel.Poles)
        poles_in_the_map_ids = {pole.id for pole in poles_in_the_map}
        self.world.enable_environment_objects(poles_in_the_map_ids, False)
        # get map information from carla
        self.map = self.world.get_map()
`

Ahmed-Radwan094, 2024-05-23:
Is there any additional information I can provide?

Blyron, 2024-08-02:
Logs on the crash. Sorry for being late, we are busy with ue5 migration

closed 2024-09-04 by Blyron