carla-simulator/carla #7687: Unreal Editor Fatal Error when using digital twin in Carla 0.9.15

lfmci, 2024-05-22:


Blyron, 2024-05-22:
We need more info, platform, version or branch, and more

lfmci, 2024-05-22:
Windows 10, version 0.9.15.
when following the digital twin tutorial and the onroad editor, the engine returns a fatal error after inputting a url from Open Street Maps and clicking generate map

Blyron, 2024-05-22:
Try latest dev version, it should be fixed

lfmci, 2024-05-22:
the version used was built from source only yesterday

Blyron, 2024-05-22:
Which is the url you are trying? Which is the crash log?

lfmci, 2024-05-22:
[CarlaUE4_4.log](https://github.com/carla-simulator/carla/files/15405778/CarlaUE4_4.log)
this was the crash log from one of the crashes

Blyron, 2024-05-23:
That log does not show any crash. I suggest you two things.
First MapName and osmname must be without any spaces. Otherwise it wont' work fine.
Second. Delete Saved folder inside Unreal/CarlaUE4/ and execute again. Saved folder will be autogenerated share it with me

lfmci, 2024-05-23:
[UE4CC-Windows-608820F54C89A65156178A8E5635797B_0000.zip](https://github.com/carla-simulator/carla/files/15413774/UE4CC-Windows-608820F54C89A65156178A8E5635797B_0000.zip)

lfmci, 2024-05-23:
that was from another attempt this morning

Blyron, 2024-05-24:
There does not appear any crash issues.  I feel like you are not in latest dev version of code. I got it working right now. Or maybe your PC specs are not good enough to generate big map

lfmci, 2024-05-24:
Ok, do I need to do a full rebuild from scratch to be able to access the latest dev code or is there a way to update?

Blyron, 2024-05-24:
Go to repo folder with console and do a git pull
then, git clean -fdx and then make launch

GOOGOL666, 2024-06-11:
> There does not appear any crash issues. I feel like you are not in latest dev version of code. I got it working right now. Or maybe your PC specs are not good enough to generate big map

Me too, I manually selected a small map,but crash when config. This is the carla-dev branch, need help, thanks!
[image]

berberlol, 2024-07-04:
> > There does not appear any crash issues. I feel like you are not in latest dev version of code. I got it working right now. Or maybe your PC specs are not good enough to generate big map
> 
> Me too, I manually selected a small map,but crash when config. This is the carla-dev branch, need help, thanks! ![Screenshot from 2024-06-11 21-18-17](https://private-user-images.githubusercontent.com/75571004/338600148-a10bd671-49e2-488e-87e5-64dc9e7ce9af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwNzM3NDQsIm5iZiI6MTcyMDA3MzQ0NCwicGF0aCI6Ii83NTU3MTAwNC8zMzg2MDAxNDgtYTEwYmQ2NzEtNDllMi00ODhlLTg3ZTUtNjRkYzllN2NlOWFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDA2MTA0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYwYWUzMjQyYzE4MGEzMDU3Y2JmZWQ2ZTJiZjk1YzgxNmZiYjNlNjM0MTA1NjcwNmNhYWE4NTA2NWRkODZjNDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rcH2ZlgiqAMkFImS1i2CbKkg2dU-blDOCrbLrBOYwhs)

same way and same question with u

EvilQQQ, 2024-12-05:
Excuse me, I would like to know what the best system configuration is, such as memory, CPU, and GPU. @Blyron

EvilQQQ, 2024-12-06:
> Thanks for that, The map size that I was trying to generate wasn't that big, maybe 20x20m but I will try again with the new code

Hi，I want to know if it worked in the end. I've encountered the same problem as well, with exactly the same log.

Trans-cending, 2024-12-21:
same problem