carla-simulator/carla #7692: Error with "make launch"

JDIRK3, 2024-05-22:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04

Problem you have experienced: When I try to start carla with "make launch" after successfully do "make PythonAPI", the unreal engine editor starts, but the progress bar stops at 71%. 
The following errors occure:
- [2024.05.22-13.25.05:351][  0]LogAssetRegistry: Error: Package ../../../../carla/Unreal/CarlaUE4/Content/Carla/Maps/TestMaps/EmptyMap.umap has malformed tag (and many similar errors)

- [2024.05.22-13.25.24:511][  0]LogLinker: Warning: The file '../../../../carla/Unreal/CarlaUE4/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset' contains unrecognizable data, check that it is of the expected type.
[2024.05.22-13.25.24:511][  0]LogLinker: Warning: The file '../../../../carla/Unreal/CarlaUE4/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset' contains unrecognizable data, check that it is of the expected type.
[2024.05.22-13.25.24:511][  0]LogUObjectGlobals: Warning: Failed to find object 'Blueprint /Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
[2024.05.22-13.25.24:511][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
[2024.05.22-13.25.24:511][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 

Steps to reproduce: I followed the introduction https://carla.readthedocs.io/en/latest/build_linux/ with additional steps provided by other issues. 

Other information (documentation you consulted, workarounds you tried): search through the other issues (in some somebody have/had the same issue in context of another issue, but no solution was provided and the issue was closed); tried ./Update.sh but got the message the content was on the newest version

ChaiJohn216, 2024-05-24:
same problem,is it solved?

JDIRK3, 2024-05-24:
No, I will post the solution, if I found one.

JDIRK3, 2024-05-24:
Maybe as addition on the error messages:

I got the following Crash report from the Unreal Engine:

libUE4Editor-Carla.so!ULoadAssetMaterialsCommandlet::ULoadAssetMaterialsCommandlet() [/home/vtd/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/LoadAssetMaterialsCommandlet.cpp:38]
libUE4Editor-CoreUObject.so!UClass::CreateDefaultObject() [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:3707]
libUE4Editor-CoreUObject.so!ProcessNewlyLoadedUObjects(FName, bool) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectBase.cpp:996]
libUE4Editor-CoreUObject.so!TBaseStaticDelegateInstance<void (FName, bool), FDefaultDelegateUserPolicy>::ExecuteIfSafe(FName, bool) const [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:729]
libUE4Editor-Core.so!TMulticastDelegate<void (FName, bool), FDefaultDelegateUserPolicy>::Broadcast(FName, bool) const [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:955]
libUE4Editor-Core.so!FModuleManager::LoadModuleWithFailureReason(FName, EModuleLoadResult&) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Modules/ModuleManager.cpp:516]
libUE4Editor-Projects.so!FModuleDescriptor::LoadModulesForPhase(ELoadingPhase::Type, TArray<FModuleDescriptor, TSizedDefaultAllocator<32> > const&, TMap<FName, EModuleLoadResult, FDefaultSetAllocator, TDefaultMapHashableKeyFuncs<FName, EModuleLoadResult, false> >&) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/ModuleDescriptor.cpp:560]
libUE4Editor-Projects.so!FPluginManager::TryLoadModulesForPlugin(FPlugin const&, ELoadingPhase::Type) const [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1293]
libUE4Editor-Projects.so!FPluginManager::LoadModulesForEnabledPlugins(ELoadingPhase::Type) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1369]
UE4Editor!FEngineLoop::LoadStartupModules() [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3808]
UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3198]
UE4Editor!GuardedMain(char16_t const*) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/vtd/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xf2)
UE4Editor!_start()

Blyron, 2024-05-24:
You have not the content available. that's why it does not work

JDIRK3, 2024-05-24:
Ok, but i extracted the input of the content.tar.gz into the content folder under "/carla/Unreal/CarlaUE4/Content/Carla". I used the newest version of the content (20231108 or something like that). What did I do wrong? 

The ../Content/Carla has following data inside:
- Blueprints
- Carla_0.9.15
- Config
- ExportedMaps
- HDMaps
- hooks
- HoudiniEngine
- Jenkinsfile
- LICENSE
- Makefile
- Maps
- road_xodr
- Static

JDIRK3, 2024-05-24:
I am not sure, but maybe I found the issue. Could it be, that Content.tar.gz needs to be extracted into /carla/Content?

For clarification before I meant with "Content.tar.gz" the file that needed to be downloaded and "Content" was just a placeholder for the Content-ID it's called, if I remember right

laqangela, 2024-05-25:
> I am not sure, but maybe I found the issue. Could it be, that Content.tar.gz needs to be extracted into /carla/Content?
>
> For clarification before I meant with "Content.tar.gz" the file that needed [...]

After update.sh is executed, the “content.tar.gz ” file package will be automatically extracted to the carla/content directory.  Have you solved the problem “ Signal 11 caught.” now? 🥺

JDIRK3, 2024-05-28:
> After update.sh is executed, the “content.tar.gz ” file package will be automatically extracted to the carla/content directory. Have you solved the problem “ Signal 11 caught.” now? 🥺

I installed two times content.tar.gz in different versions and at different places. After deleting both of them and running update.sh it worked. No signal 11 left.

JDIRK3, 2024-06-03:
Solution was to completly everything new. So something must gone wrong during installation would be my guess.

closed 2024-06-03 by JDIRK3

johnren-code, 2024-08-01:
I don't fully understand what you mean. What does the specific solution refer to?

liaopuyun, 2024-09-20:
> I don't fully understand what you mean. What does the specific solution refer to?

I suppose you need to re-run the ./Update.sh correctly, so that we have a complete Content folder