carla-simulator/carla #7701: Need a way to get the current door status of a vehicle

cmpute, 2024-05-24:
As described in title.

PabloVD, 2024-05-27:
You can close and open doors of a vehicle using the methods `open_door` and `close_door`, as described [in the documentation](https://carla.readthedocs.io/en/latest/python_api/#methods_55). Currently there is not a method to get the current state of a door in the Python API, but you can easily access to that information using a boolean, as is done in the [manual control example](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/manual_control.py).

PabloVD, 2024-06-07:
Closing due to inactivity. Please feel free to reopen if the issue persists.

Kait0, 2025-01-30:
@PabloVD Please reopen this issue (I do not have permission). Not being able to query the door state is an issue in cases where the door state is not set by your code.
This is for example the case with the CARLA scenario runner, where the door state is set in the scenario runner that users of the CARLA leaderboard should not access.
Another problem is the replay system of CARLA. Since the door state is not logged, the replay is missing the door behavior.

I think it should be possible to query the door state of a vehicle via the CARLA API. Logging this info should not create any meaningful overhead.

closed 2026-11-13 by Blyron