carla-simulator/carla #7704: Controlling Pedestrians with Motion Capture Devices in Carla

Ikaros520520, 2024-05-27:
Hello, everyone！

I'm new to using Carla.

I have a motion-capture set on hand. I want to use this set to control the pedestrians in Carla. As shown in the following documentation (https://arxiv.org/pdf/2206.00337). I didn't find anything in Carla's documentation that would help me. There is no detail in the above paper on how the motion capture device controls pedestrians. Whether it's a relevant help document, a specific step-by-step procedure, or a similar example, please let me know.
Much appreciated!

CARLA version:0.9.12
Platform/OS:Window10
Motion Capture Equipment:Perception Neuron Studio(https://neuronmocap.com/pages/perception-neuron-studio-system)

GoodarzMehr, 2024-05-28:
Hi,

You can take a look at the [WalkerBoneControl](https://carla.readthedocs.io/en/0.9.12/python_api/#carla.WalkerBoneControl) class and the [Walker](https://carla.readthedocs.io/en/0.9.12/python_api/#carlawalker) class' `apply_control` method for your specific version. I think newer versions have some expanded control which you can also check out in the documentation.

DnDare, 2024-05-31:
Hi, 
Thanks GoodarzMehr, 
We don't have at Carla any documentation on adding real-time motion-capture. You'll have to research by yourself , or ask the authors of the paper you mentioned. 
Sorry about that.

There is an [issue](https://github.com/carla-simulator/carla/issues/7457#issuecomment-2077647526) related to pedestrians, explains a little about animgraphs, and how we made it in Carla.

If you review this [video](https://www.youtube.com/watch?v=jRyq5uPC5UY), you might get some insights on how to connect MOCAP to the simulation.

Hope it helps,

closed 2024-06-07 by DnDare

Ikaros520520, 2024-06-09:
Thanks for the help!
I finally found out how to drive the UE character models through the official help documentation for the motion capture device(Perception Neuron Studio).