carla-simulator/carla #7706: Fake Interior Building Shader

Madecu, 2024-05-27:
Create new fake interior shadr that simulates glass in front of it for the new nanite buildings

DnDare, 2024-07-30:
Custom new FakeInterior textures, generated with StableDiffusion.
And  Bottom Store building WIP

Interiors generated:  Convenience Store, Kitchen room, apartmen room, coffe shop

[image]
[image]
[image]
[image]
[image]

DnDare, 2024-07-30:
Those that give a better result, are the limited to a Cube space,  deeper rooms or alleys, can give a worse perspective effect

closed 2024-09-17 by Madecu

DnDare, 2024-09-18:
[image]
[image]
HDRI  interior  in Door and Corner nodes ,  bottomStore building