carla-simulator/carla #7710: Carla server crashed.Signal 11 caught. Malloc Size=65538 LargeMemoryPoolOffset=65554

veinlove55, 2024-05-27:
CARLA version: release-0.9.15

Platform/OS: Ubuntu 20.04

Problem you have experienced: 
ubuntu@ubuntu-MS-7D94:~/carla-release-0.9.15$ ./CarlaUE4.sh -RenderOffScreen
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

What you expected to happen:
No crush happens.

Steps to reproduce:
client in C++ built.
town_name = "/Game/Carla/Maps/Town10HD_Opt";

Anyone met the same crush?

lcsc0, 2024-07-04:
I got the same issue

teimuri, 2024-09-30:
I also get the same error

EstelleY0, 2025-01-21:
I also also get the same error ;(