carla-simulator/carla #7711: The output point cloud is referenced to what coordinate system, relative to the vehicle itself or the LiDAR?

huangchunan, 2024-05-27:
Platform : CARLA 0.9.15
Questions ：
I am currently researching how to generate point cloud data using Carla. I met three questions.
1）When I use sensor.get_location(), it always show (0,0,0). I don't know where the LiDAR is attached to the vehicle. Any way to show the lidar sensor in Carla?
2）The output point cloud is referenced to what coordinate system, relative to the vehicle itself or the LiDAR?
3）Does weather affect the LiDAR point cloud?
Any answer is welcome!

    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        world = self._parent.get_world()
        self.debug = world.debug
        lidar_bp = world.get_blueprint_library().find('sensor.lidar.ray_cast_semantic')
        lidar_bp.set_attribute('channels', str(64))
        lidar_bp.set_attribute('points_per_second', str(1152000))
        lidar_bp.set_attribute('rotation_frequency', str(10))
        lidar_bp.set_attribute('range', str(15))
        lidar_bp.set_attribute('upper_fov', str(15))
        lidar_bp.set_attribute('lower_fov', str(-25))
        # lidar_bp.set_attribute('sensor_tick', str(0.1))

        self.sensor = world.spawn_actor(
            lidar_bp,
            carla.Transform(
                carla.Location(z=1.8372)),
            attach_to=self._parent)
        # time.sleep(5)
        if self.sensor.parent:
            print(f"Sensor is attached to: {self.sensor.parent.type_id}")
        print(f"lidar_sensor:{self.sensor.get_location()}")
        output_path = "/tmp/carla_lidar"
        self.sensor.listen(
            lambda point_cloud: \
                    point_cloud.save_to_disk(os.path.join(output_path, '%06d.ply' % point_cloud.frame)))

GoodarzMehr, 2024-05-28:
Hi, as far as I know, 2) relative to the lidar, 3) no, weather is only visually represented (only affects the rgb camera).

johnren-code, 2024-12-03:
Hello, I want to know what is the relative position of this coordinate system to the orientation of the vehicle?

GoodarzMehr, 2024-12-03:
Hi, whatever you set it to when creating the lidar (if attaching it to the vehicle).