carla-simulator/carla #7717: Agent unresponsive for first 50 ticks in PythonAPI/examples/automatic_control.py

DP1997, 2024-05-28:
CARLA version: 0.9.12
Platform/OS: Windows 10
Problem you have experienced: The agent (ego-vehicle) is unresponsive for the first 50 ticks of the server, i.e. the agent doesn't move even though the relevant carla.VehicleControl() object has throttle set to 0.75.
What you expected to happen: I expect the agent to immediately respond with the tick that applies the current carla.VehicleControl().
Steps to reproduce: Download Carla 0.9.12, edit automatic_control.py to print current step & speed, then execute.
Other information (documentation you consulted, workarounds you tried): Apparently the number of ticks required for the agent to actually start executing the supplied commands through carla.VehicleControl() objects depends on the fixed_delta_seconds parameter. 

[image]

[image]
[image]

PabloVD, 2024-05-31:
Hi, have you started the server before running the automatic control script?
You have to run the CarlaUE4 executable first to start the Carla simulation, and then run the automatic control script.

DP1997, 2024-06-02:
Hi @PabloVD, thanks for your reply!

Yes, the CARLA server is already running before I am executing the python script.

At the moment my temporary solution is to supply dummy VehicleControls with throttle = 1 and tick the server until the ego vehicle starts to actually move. I then switch to setting VehicleControl with brake = 1 and tick until the car comes to a complete stop again. Afterwards i reset the car to its initial starting position.
After that any VehicleControl command is immediately applied.

glopezdiest, 2024-06-07:
@DP1997 This might be because of the vehicle physics. 

Not sure why this is the case, but from testing, we saw that vehicles take around 0.5 seconds to start moving the first time after being spawned. We never really found out why, but it might be due to PhysX engaging the motor? If I recall correctly, one way to remove it was to set the _manual_gear_shift_ to True, and the _gear_ to 1 during one tick.

Closing the issue, but feel free to reopen it if it doesn't work

closed 2024-06-07 by glopezdiest