carla-simulator/carla #7721: How to publish the Carla project compiled from source code into an executable file

yuxinhi, 2024-05-29:
CARLA Version：0.9.14
OS：Ubuntu 20.04

When I run "make package" and "make CarlaUE4Editor," the system return errors.
so, How to publish the Carla project compiled from source code on a Linux system into an executable file, without the need to open the editor every time the project starts？

thanks for your help

PatrickPromitzer, 2024-05-29:
Hi,
It dependents of the error.

Can you post the output of the error message and post it here?
The error message can be in between the whole command line output.

PabloVD, 2024-06-21:
You have to follow [these steps](https://carla.readthedocs.io/en/latest/build_linux/) to build CARLA from source. After compiling with `make launch`, you can use `make package` to create the executable package. As @PatrickPromitzer suggests, if you share the errors you get we could have more information on how to solve the problem.

closed 2025-02-19 by Blyron