carla-simulator/carla #7722: error when running make osmrenderer in ubuntu2 2.04

lfmci, 2024-05-29:
Carla version: 0.9.15
OS:Ubuntu 22.04
exit with error after reaching 70% on make osmrenderer command. code is below.
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `void absl::lts_20230802::log_internal::LogMessage::CopyToEncodedBuffer<(absl::lts_20230802::log_internal::LogMessage::StringType)0>(std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage::OstreamView::stream()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::GetAppendBufferSlowPath(unsigned long, unsigned long, unsigned long)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage& absl::lts_20230802::log_internal::LogMessage::operator<< <unsigned int, 0>(unsigned int const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage::operator<<(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::raw_log_internal::internal_log_function[abi:cxx11]'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::AppendPrecise(std::basic_string_view<char, std::char_traits<char> >, absl::lts_20230802::cord_internal::CordzUpdateTracker::MethodIdentifier)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::base_internal::ThrowStdOutOfRange(char const*)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessageFatal::LogMessageFatal(char const*, int)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::operator=(std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage& absl::lts_20230802::log_internal::LogMessage::operator<< <unsigned short, 0>(unsigned short const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::hash_internal::MixingHashState::kSeed'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::GetFlatAux(absl::lts_20230802::cord_internal::CordRep*, std::basic_string_view<char, std::char_traits<char> >*)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::hash_internal::MixingHashState::CombineLargeContiguousImpl64(unsigned long, unsigned char const*, unsigned long)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessageFatal::~LogMessageFatal()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::RemoveSuffix(unsigned long)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::CheckOpMessageBuilder::ForVar2()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Mutex::~Mutex()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::strings_internal::StringifySink::Append(std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage::LogMessage(char const*, int, absl::lts_20230802::log_internal::LogMessage::ErrorTag)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage& absl::lts_20230802::log_internal::LogMessage::operator<< <void*, 0>(void* const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::Append(absl::lts_20230802::Cord&&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::CheckOpMessageBuilder::NewString[abi:cxx11]()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::InlineRep::AppendTree(absl::lts_20230802::cord_internal::CordRep*, absl::lts_20230802::cord_internal::CordzUpdateTracker::MethodIdentifier)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::InlineRep::UnrefTree()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::strings_internal::CatPieces[abi:cxx11](std::initializer_list<std::basic_string_view<char, std::char_traits<char> > >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessageFatal::LogMessageFatal(char const*, int, std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::numbers_internal::kHexTable'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::container_internal::DropDeletesWithoutResize(absl::lts_20230802::container_internal::CommonFields&, absl::lts_20230802::container_internal::PolicyFunctions const&, void*)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::DestroyCordSlow()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::StrCat[abi:cxx11](absl::lts_20230802::AlphaNum const&, absl::lts_20230802::AlphaNum const&, absl::lts_20230802::AlphaNum const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage::LogMessage(char const*, int, absl::lts_20230802::log_internal::LogMessage::WarningTag)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::StrCat[abi:cxx11](absl::lts_20230802::AlphaNum const&, absl::lts_20230802::AlphaNum const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Mutex::Unlock()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::MakeCheckOpValueString(std::ostream&, unsigned char)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Mutex::Lock()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `utf8_range::IsStructurallyValid(std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::MakeCheckOpValueString(std::ostream&, void const*)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::Append(absl::lts_20230802::Cord const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage& absl::lts_20230802::log_internal::LogMessage::operator<< <int, 0>(int const&)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::kCharNull'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::Clear()'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::Cord::ChunkIterator::AdvanceAndReadBytes(unsigned long)'
/usr/bin/ld: ../libosmscout-import/libosmscout_import.so.1.1.1: undefined reference to `absl::lts_20230802::log_internal::LogMessage::OstreamView::~OstreamView()'
[... cut ...]
make[3]: *** [Import/CMakeFiles/Import.dir/build.make:106: Import/Import-1.1.1] Error 1
make[3]: Leaving directory '/home/ctrlrm-adm/carla/Build/libosmscout-build'
make[2]: *** [CMakeFiles/Makefile2:1074: Import/CMakeFiles/Import.dir/all] Error 2
make[2]: Leaving directory '/home/ctrlrm-adm/carla/Build/libosmscout-build'
make[1]: *** [Makefile:146: all] Error 2
make[1]: Leaving directory '/home/ctrlrm-adm/carla/Build/libosmscout-build'
make: *** [Util/BuildTools/Linux.mk:160: osmrenderer] Error 2
any help on this issue would be greatly appreciated

bg-sumu, 2024-09-03:
I also have the same issue

2409Kevin, 2024-12-10:
Have you solved it? I encountered the problem as well