carla-simulator/carla #7723: Memory usage increasing indefinitely while CarlaUE4 is minimized

apolak, 2024-05-29:
**CARLA version:** 0.9.15

**Platform/OS:** Ubuntu 20.04

**Problem you have experienced:** After minimizing the CarlaUE4 window, memory usage starts to increase indefinitely. It returns to normal as soon as the window is restored. See the screenshot below, where the previously minimized window has been restored near the 10 second mark.

[image]

**What you expected to happen:** Memory usage is normal when the CarlaUE4 window is minimized.

**Steps to reproduce:**

1. Start CarlaUE4 in on-screen mode:
    ```bash
    ./CarlaUE4.sh
    ```

2. Start the `manual_control.py` script in another terminal:
    ```bash
    cd ./PythonAPI/examples
    python3 manual_control.py
    ```

3. Minimize the CarlaUE4 window.
4. Monitor the memory usage in another terminal:
    ```bash
    htop -p $(pgrep -f CarlaUE4-Linux-Shipping)
    ```

**Other information:**

1. I was only able to reproduce this issue while running a Python API client script.
2. I couldn't reproduce this issue with the `no_rendering_mode.py` script—possibly a camera sensor issue?

Blyron, 2024-05-30:
Hello! I have not been able to repro it.
Which commands have you introduced in the Python client? have you enabled the recorder o maybe enabled de save images function for camera sensors? 
Is this prebuilt version or it is build from souce?

apolak, 2024-05-30:
@Blyron This is the release version with additional maps installed using `ImportAssets.sh`. Both packages were downloaded using links from the [release page](https://github.com/carla-simulator/carla/releases/tag/0.9.15). I'm using the original Python scripts with no modifications.

I now recall that I've never seen this issue before installing the additional maps package. Maybe this is related?

One thing to note is that to reproduce the issue, I don't need to load any of the additional maps or interact with the Python client. It's enough to start CarlaUE4 and the Python client with no command line options, like described above, then minimize the CarlaUE4 window.

I'm using the pre-installed Python 3.8 version with a virtual environment created like below:

```bash
sudo apt-get install python3-venv
cd ./PythonAPI/examples
python3 -m venv ~/carla-venv
source ~/carla-venv/bin/activate
pip install --upgrade pip
pip install -r requirements.txt
pip install carla==0.9.15
```

The machine is an AWS-hosted `g4dn.xlarge` instance created from the AWS-provided Ubuntu Server 20 LTS x86 image to which I connect using NICE DCV.

Blyron, 2024-05-31:
Which Vulkan Drivers is using your machine?

apolak, 2024-05-31:
@Blyron It's the NVIDIA GRID driver v550.54.17, see the output of `vulkaninfo` below. I tried NVIDIA Tesla T4 and NVIDIA L4 GPUs and was able to reproduce this issue on both.

<details>
<summary>vulkaninfo</summary>

```console
==========
VULKANINFO
==========

Vulkan Instance Version: 1.2.131

Instance Extensions: count = 19
====================
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 10
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_EXT_swapchain_colorspace            : extension revision 4
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 23
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 2
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_wayland_surface                 : extension revision 6
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 4
=======
VK_LAYER_LUNARG_standard_validation (LunarG Standard Validation Layer) Vulkan version 1.0.131, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id 	: 0 (Tesla T4)
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.2.73, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id 	: 0 (Tesla T4)
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.1.73, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id 	: 0 (Tesla T4)
		Layer-Device Extensions: count = 0

VK_LAYER_NV_optimus (NVIDIA Optimus layer) Vulkan version 1.3.277, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id 	: 0 (Tesla T4)
		Layer-Device Extensions: count = 0

Presentable Surfaces:
=====================
GPU id : 0 (Tesla T4):
	Surface types: count = 2
		VK_KHR_xcb_surface
		VK_KHR_xlib_surface
	Formats: count = 2
		SurfaceFormat[0]:
			format = FORMAT_B8G8R8A8_UNORM
			colorSpace = COLOR_SPACE_SRGB_NONLINEAR_KHR
[... 1231 lines cut ...]
	separateDepthStencilLayouts                        = true
	hostQueryReset                                     = true
	timelineSemaphore                                  = true
	bufferDeviceAddress                                = true
	bufferDeviceAddressCaptureReplay                   = true
	bufferDeviceAddressMultiDevice                     = true
	vulkanMemoryModel                                  = true
	vulkanMemoryModelDeviceScope                       = true
	vulkanMemoryModelAvailabilityVisibilityChains      = true
	shaderOutputViewportIndex                          = true
	shaderOutputLayer                                  = true
	subgroupBroadcastDynamicId                         = true

VkPhysicalDeviceVulkanMemoryModelFeatures:
------------------------------------------
	vulkanMemoryModel                             = true
	vulkanMemoryModelDeviceScope                  = true
	vulkanMemoryModelAvailabilityVisibilityChains = true

VkPhysicalDeviceYcbcrImageArraysFeaturesEXT:
--------------------------------------------
	ycbcrImageArrays = true
```

</details>

Blyron, 2024-05-31:
We have repro it. It seems is UE issue
https://forums.unrealengine.com/t/when-minimized-i-have-a-memory-leak-in-packaged-game-when-only-using-vulcan-or-dx12-any-ideas/1429649
We are looking at it

Blyron, 2024-05-31:
Which is the FPS of your client while server minimized?

Blyron, 2024-05-31:
Have you tried with -RenderOffScreen?

apolak, 2024-05-31:
> Have you tried with -RenderOffScreen?

@Blyron If I start CarlaUE4 if off-screen mode, after starting the Python client I can also see memory usage constantly increasing, but at a slower rate, see the screenshot below. I started the Python client around the 8:55 mark. At the 9:35 mark, the percentage of memory used has increased from 28% to 32%. This trend isn't visible if I start CarlaUE4 in on-screen mode (not minimized).
[image]

Blyron, 2024-05-31:
Okay. We got it, we are able to repro it. But we do not think it would be quick fix. Seems to be related with Unreal itself. We need to dig into that

Blyron, 2024-11-06:
It should be fixed now. Can you confirm it?

purewater0901, 2024-11-21:
@Blyron Hello Byron. Thank you for maintaining the Carla simulator.

I also had the same issues with Carla 0.9.15, and I found this thread. In order to confirm that the problem has fixed, should I build and install the latest Carla?

Blyron, 2024-11-21:
Yeah, in ue4-dev should be fixed

purewater0901, 2024-11-21:
Got it. I will reinstall the Carla and check it later.

closed 2025-02-19 by Blyron