carla-simulator/carla #7735: Shortest Route between point_a and point_b

mariashaima, 2024-05-31:
Could someone please guide me on how to identify the shortest route between point_a and point_b using the provided code as reference? 
the longest route between point_a and point_b is already given in the code. Any suggestions or examples would be greatly appreciated. 
Thank you!
code [https://github.com/mariashaima/route_carla/commit/f7502c9338142b88f0c5394ee1636656a8e3e4c7]

glopezdiest, 2024-06-07:
Hey @mariashaima, the link you provided is not working.

But overall, you could use the [GlobalRoutePlanner](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/carla/agents/navigation/global_route_planner.py), which has the `trace_route` function to get the waypoints of that route

mariashaima, 2024-06-07:
[https://github.com/mariashaima/route_carla/blob/main/short_path.py](url)

mariashaima, 2024-06-07:
> Hey @mariashaima, the link you provided is not working.
> 
> But overall, you could use the [GlobalRoutePlanner](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/carla/agents/navigation/global_route_planner.py), which has the `trace_route` function to get the waypoints of that route

check out this new link : [https://github.com/mariashaima/route_carla/blob/main/short_path.py](url)

mariashaima, 2024-06-07:
hey @glopezdiest  please check this new link?
[https://github.com/mariashaima/route_carla/blob/main/short_path.py](url)

mariashaima, 2024-06-07:
hey  @glopezdiest ,
check out this link ? [https://github.com/mariashaima/route_carla]

mariashaima, 2024-06-07:
@glopezdiest 
check out this link?[https://github.com/mariashaima/route_carla](url)

mariashaima, 2024-06-07:
[https://github.com/carla-simulator/carla/issues](url)

mariashaima, 2024-06-07:
[https://github.com/mariashaima/route_carla.git](url)

closed 2024-09-30 by MarcelPiNacy-CVC