carla-simulator/carla #7736: Question About Carla’s Synchronous Mode

YuqiHuai, 2024-06-02:
Hi Carla community!

I am trying to understand Carla's synchronous mode by reading [this](https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/#client-server-synchrony) documentation.

> By default, CARLA runs in asynchronous mode. The server runs the simulation as fast as possible, without waiting for the client. On synchronous mode, the server waits for a client tick, a "ready to go" message, before updating to the following simulation step. 
...
> The synchronous mode becomes specially relevant with slow client applications, and when synchrony between different elements, such as sensors, is needed. If the client is too slow and the server does not wait, there will be an overflow of information. The client will not be able to manage everything, and it will be lost or mixed. On a similar tune, with many sensors and asynchrony, it would be impossible to know if all the sensors are using data from the same moment in the simulation. 

Q1: What exactly does `client` refer to here? If I connect Carla with an ADS (e.g., Autoware/Apollo), does that ADS become `client`? If the ADS connects to Carla via a bridge, does `client` refer to the bridge or the ADS?

Q2: Under `synchronomous mode`, does the ADS's clock gets automatically synchronized with Carla's clock without additional setup? ADS here specificaly refers to Autoware/Apollo.

closed 2024-06-07 by glopezdiest

YuqiHuai, 2024-06-07:
Hi @glopezdiest , thank you for your detailed answer! This is very helpful to me!
Just to confirm with you again regarding Q2, using Carla's synchronized mode does not automatically guarantee synchronization of Carla's simulation clock with the ADS's clock, and I have to verify the implementation of the "bridge" to see if any synchronization is done, would this understanding be correct?

glopezdiest, 2024-06-17:
Yes