carla-simulator/carla #7737: Build carla ue5-dev on Windows10 failed

lryyyyy, 2024-06-02:
Hi Carla community!

I'm trying to build from source by ue5-dev branch, follows by [Setup.bat](https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.9bekxmmygkbc), but suffer a build error when run "**cmake --build Build --target launch**"
Here is error screenshots:
[image]

Total log is as follows:

D:\Simulators\carla>cmake --build Build --target launch
[0/3] C:\Windows\system32\cmd.exe /C "cd /D D:\Simulators\carla\Build\Unreal && D:\Simulators\UnrealEng...Win64 Development -project=D:/Simulators/carla/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscwUsing bundled DotNet SDK version: 6.0.302
Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" CarlaUnrealEditor Win64 Development -project=D:/Simulators/carla/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw
Log file: D:\Simulators\UnrealEngine5_carla\Engine\Programs\UnrealBuildTool\Log.txt
Using 'git status' to determine working set for adaptive non-unity build (D:\Simulators\UnrealEngine5_carla).
Waiting for 'git status' command to complete
Building CarlaUnrealEditor...
Using Visual Studio 2022 14.40.33811 toolchain (D:\ProgramFiles\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.40.33807) and Windows 10.0.18362.0 SDK (D:\Windows Kits\10).
Determining max actions to execute in parallel (6 physical cores, 12 logical cores)
  Executing up to 6 processes, one per physical core
  Requested 1.5 GB memory per action, 7.49 GB available: limiting max parallel actions to 4
------ Building 7 action(s) started ------
[1/7] Compile [x64] Module.Carla.10.cpp
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Recorder\CarlaRecorderQuery.cpp(57): warning C4996: 'localtime': This function or variable may be unsafe. Consider using localtime_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\DVSCamera.cpp(133): warning C4996: 'UObjectBaseUtility::IsPendingKill': IsPendingKill() should no longer be used. Use IsValid(Object), IsValidChecked(Object) or GetValid(Object) instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\ImageUtil.cpp(212): warning C4996: 'UTexture::Resource': Use GetResource() / SetResource() accessors instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.cpp(30): warning C4996: 'UTexture::Resource': Use GetResource() / SetResource() accessors instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.cpp(56): warning C4996: 'FRHICommandListImmediate::GetRenderQueryResult': Use the global scope RHIGetRenderQueryResult function. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\SceneCaptureSensor.cpp(567): warning C4996: 'UObjectBaseUtility::IsPendingKill': IsPendingKill() should no longer be used. Use IsValid(Object), IsValidChecked(Object) or GetValid(Object) instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Server\CarlaServer.cpp(490): warning C4996: 'ACarlaGameModeBase::GetNamesOfAllActors': This command no longer makes sense since the apply_textures API relies on IDs.
D:\Simulators\carla\LibCarla\source\carla\AtomicSharedPtr.h(28): warning C4996: 'std::atomic_store_explicit': warning STL4029: std::atomic_*() overloads for shared_ptr are deprecated in C++20. The shared_ptr specialization of std::atomic provides superior functionality. You can define _SILENCE_CXX20_OLD_SHARED_PTR_ATOMIC_SUPPORT_DEPRECATION_WARNING or _SILENCE_ALL_CXX20_DEPRECATION_WARNINGS to suppress this warning.
D:\Simulators\carla\LibCarla\source\carla\AtomicSharedPtr.h(28): note: 模板实例化上下文(最早的实例化上下文)为
D:\Simulators\carla\LibCarla\source\carla\streaming\detail\MultiStreamState.h(137): note: 查看对正在编译的 类 模板 实例化“carla::AtomicSharedPtr<carla::streaming::detail::Session>”的引用
D:\Simulators\carla\LibCarla\source\carla\AtomicSharedPtr.h(27): note: 在编译 类 模板 成员函数“void carla::AtomicSharedPtr<carla::streaming::detail::Session>::store(std::shared_ptr<_Ty>) noexcept”时
        with
        [
            _Ty=carla::streaming::detail::tcp::ServerSession
        ]
D:\Simulators\carla\LibCarla\source\carla\streaming\detail\MultiStreamState.h(87): note: 请参阅 "carla::streaming::detail::MultiStreamState::ConnectSession" 中对 "carla::AtomicSharedPtr<carla::streaming::detail::Session>::store" 的第一个引用
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(102): warning C4996: 'UObjectBaseUtility::IsPendingKill': IsPendingKill() should no longer be used. Use IsValid(Object), IsValidChecked(Object) or GetValid(Object) instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(102): note: 模板实例化上下文(最早的实例化上下文)为
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\OpticalFlowCamera.cpp(53): note: 查看对正在编译的函数 模板 实例化“void FPixelReader::SendPixelsInRenderThread<AOpticalFlowCamera,float>(TSensor &,bool,std::function<TArray<float,FDefaultAllocator> (void *,uint32)>)”的引用
        with
        [
            TSensor=AOpticalFlowCamera
        ]
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(126): warning C4996: 'UObjectBaseUtility::IsPendingKill': IsPendingKill() should no longer be used. Use IsValid(Object), IsValidChecked(Object) or GetValid(Object) instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(126): note: 模板实例化上下文(最早的实例化上下文)为
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\OpticalFlowCamera.cpp(53): note: 查看对正在编译的函数 模板 实例化“void FPixelReader::SendPixelsInRenderThread<AOpticalFlowCamera,float>(TSensor &,bool,std::function<TArray<float,FDefaultAllocator> (void *,uint32)>)”的引用
        with
        [
            TSensor=AOpticalFlowCamera
        ]
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(113): note: 查看对正在编译的函数 模板 实例化“void EnqueueUniqueRenderCommand<FPixelReader::SendPixelsInRenderThread::FWritePixels_SendPixelsInRenderThreadName,FPixelReader::SendPixelsInRenderThread::<lambda_1>>(LAMBDA &&)”的引用
        with
        [
            LAMBDA=FPixelReader::SendPixelsInRenderThread::<lambda_1>
        ]
D:\Simulators\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Public\RenderingThread.h(270): note: 查看对正在编译的函数 模板 实例化“auto FPixelReader::SendPixelsInRenderThread::<lambda_1>::operator ()<FRHICommandListImmediate>(_T2 &)”的引用
        with
        [
            _T2=FRHICommandListImmediate
        ]
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(234): error C2653: “get”: 不是类或命名空间名称
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(234): error C2510: “get”:“::”的左边必须是类/结构/联合
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(234): error C2065: “header_offset”: 未声明的标识符
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\PixelReader.h(120): warning C4996: 'UObjectBaseUtility::IsPendingKill': IsPendingKill() should no longer be used. Use IsValid(Object), IsValidChecked(Object) or GetValid(Object) instead. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\AsyncDataStreamImpl.h(27): error C2653: “get”: 不是类或命名空间名称
[... cut ...]
D:\Simulators\carla\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\AsyncDataStreamImpl.h(23): note: 尝试匹配参数列表“(void)”时
Total time in Parallel executor: 42.17 seconds
Total execution time: 46.40 seconds

FAILED: Unreal/CMakeFiles/carla-unreal-editor D:/Simulators/carla/Build/Unreal/CMakeFiles/carla-unreal-editor
C:\Windows\system32\cmd.exe /C "cd /D D:\Simulators\carla\Build\Unreal && D:\Simulators\UnrealEngine5_carla\Engine\Build\BatchFiles\Build.bat CarlaUnrealEditor Win64 Development -project=D:/Simulators/carla/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"
ninja: build stopped: subcommand failed.

Blyron, 2024-06-03:
Hey, try to follow this steps
https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.5piein3604b7

lryyyyy, 2024-06-09:
> Hey, try to follow this steps https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.5piein3604b7

I followed your guide, but I still encountered the same issue. May I ask if you successfully built it on Windows 10? Are there any additional environment variables or dependencies needed?

hs2022gh, 2024-06-11:
Hi, I've got the same error. 
but i finally have got succeeded by solution in this URL : https://www.reddit.com/r/unrealengine/comments/1axxe63/can_anyone_help_me_with_this_unreal_5_error/

Blyron, 2024-06-18:
> I followed your guide, but I still encountered the same issue. May I ask if you successfully built it on Windows 10? Are there any additional environment variables or dependencies needed?

People have been compalinning on Windows 10, I am using Windows 11 and everything works fine for me. Haven't had the chance to look at it in Windows 10

lryyyyy, 2024-06-27:
> People have been compalinning on Windows 10, I am using Windows 11 and everything works fine for me. Haven't had the chance to look at it in Windows 10

I have tried it again, but it still can't works on my envionment.

Blyron, 2024-07-25:
Please provide logs completely in English. Otherwise I won't be able to understand the issues. but seems like something is missing on your machine. Maybe rerun Setup.bat

zhaosiwen1949, 2024-07-26:
@Blyron I was blocked by the third step when I use this command "cmake --build Build --target launch"
----------------------------------------------------------------------------------------------------------------------------------------------
This is the whole error log:
[0/2] Re-checking globbed directories...
[0/3] C:\Windows\system32\cmd.exe /C "cd /D C:\WorkSpace\C...5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"Using bundled DotNet SDK version: 6.0.302
Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" CarlaUnrealEditor Win64 Development -project=C:/WorkSpace/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw
Log file: C:\WorkSpace\UnrealEngine5_carla\Engine\Programs\UnrealBuildTool\Log.txt
Using 'git status' to determine working set for adaptive non-unity build (C:\WorkSpace\UnrealEngine5_carla).
Waiting for 'git status' command to complete
Building CarlaUnrealEditor...
Using Visual Studio 2022 14.36.32546 toolchain (C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.36.32532) and Windows 10.0.22621.0 SDK (C:\Program Files (x86)\Windows Kits\10).
Determining max actions to execute in parallel (16 physical cores, 24 logical cores)
  Executing up to 16 processes, one per physical core
------ Building 3 action(s) started ------
[1/3] Link [x64] UnrealEditor-CarlaTools.dll
   Creating library C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.sup.lib and object C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.sup.exp
carla-server.lib(Map.cpp.obj) : error LNK2019: unresolved external symbol __std_min_element_d referenced in function "double * __cdecl std::_Min_element_unchecked<double *,struct std::less<void> >(double *,double *,struct std::less<void>)" (??$_Min_element_unchecked@PEANU?$less@X@std@@@std@@YAPEANPEAN0U?$less@X@0@@Z)
C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Binaries\Win64\UnrealEditor-CarlaTools.dll : fatal error LNK1120: 1 unresolved externals
[2/3] Link [x64] UnrealEditor-Carla.dll
   Creating library C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Intermediate\Build\Win64\x64\UnrealEditor\Development\Carla\UnrealEditor-Carla.sup.lib and object C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Intermediate\Build\Win64\x64\UnrealEditor\Development\Carla\UnrealEditor-Carla.sup.exp
carla-server.lib(Map.cpp.obj) : error LNK2019: unresolved external symbol __std_min_element_d referenced in function "double * __cdecl std::_Min_element_unchecked<double *,struct std::less<void> >(double *,double *,struct std::less<void>)" (??$_Min_element_unchecked@PEANU?$less@X@std@@@std@@YAPEANPEAN0U?$less@X@0@@Z)
C:\WorkSpace\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Binaries\Win64\UnrealEditor-Carla.dll : fatal error LNK1120: 1 unresolved externals
Total time in Parallel executor: 0.73 seconds
Total execution time: 4.04 seconds

FAILED: Unreal/CMakeFiles/carla-unreal-editor C:/WorkSpace/CarlaUE5/Build/Unreal/CMakeFiles/carla-unreal-editor
C:\Windows\system32\cmd.exe /C "cd /D C:\WorkSpace\CarlaUE5\Build\Unreal && C:\WorkSpace\UnrealEngine5_carla\Engine\Build\BatchFiles\Build.bat CarlaUnrealEditor Win64 Development -project=C:/WorkSpace/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"
ninja: build stopped: subcommand failed.
[image]
------------------------------------------------------------------------------------------------------------------------------------------
The other steps are all successful.

By the way, I also tried @hs2022gh 's method, installed 14.36 version's MSVC toolchain, but can't help.
[image]

I use Windows 10.0.22621.0 SDK and 14.36.32532 MSVC Tool.

Is there any solutions? This is blocking me for 2 weeks~

zhaosiwen1949, 2024-07-27:
From logs, it seem like there is unresolved external symbol in carla-server.lib, but I can't find it. By the way, @hs2022gh 's method is efficient, there's no error as @lryyyyy‘s, but new errors occur.

zhaosiwen1949, 2024-07-27:
Maybe it's because the different version of Windows SDK ? @Blyron can you provide the version of Windows SDK and MSVC Tools?

Blyron, 2024-07-29:
I have the versions installed by carlasetup.bat

jeromepatel, 2024-07-29:
For me the original issue by @lryyyyy was resolved with solution from @hs2022gh. Thanks for the help! I have tried to sum-it-up in another issue which occurred together (probably because of some latest update in a dependency which broke the setup script). Here's the detailed steps [mentioned in another comment](https://github.com/carla-simulator/carla/issues/7782#issuecomment-2256756786)

J160KU, 2024-07-30:
**I got a similar message in Windows 11:**

[image]

**ChatGPT comes to following conclusions regarding the first three red error message**s:

The build process has failed due to several errors. Here are some specific errors and how to address them:

    Error C2653: This error indicates that the compiler cannot find a class or namespace named get. Ensure that you have included the necessary headers and that the get function is defined in the appropriate namespace.

    Error C2065: This error indicates that index is undeclared. Check the code in AsyncDataStreamImpl.h to ensure that index is properly declared before it is used.

    Error C2665: This error indicates that the constructor for carla::Buffer cannot be called with the provided arguments. Check the arguments being passed to the constructor in AsyncDataStreamImpl.h and ensure they match the expected types.

@Blyron Please could you doublecheck, if ChatGPT's answer is correct (no hallucination)?

Blyron, 2024-07-30:
If there were those compilation issues I would have those on my end, I have no changes right now and it's compiling properly. Something is missing on vs setup. 
Try to solve by this:
https://github.com/carla-simulator/carla/issues/7782#issuecomment-2256756786

J160KU, 2024-07-30:
**Thank you for your quick answers @Blyron & @jeromepatel!** 

I could solve the red error messages with @hs2022gh's suggestion discussed at [Reddit ](https://www.reddit.com/r/unrealengine/comments/1axxe63/can_anyone_help_me_with_this_unreal_5_error/): 

"The fix was to go into _Visual Studio Installer_ and enable the option under > Modify > Individual Components > _MSVC v143 - VS 2022 C++ x64/x86 build tools (v14.36-17.6)_"

So during compiling in the build process there comes the same red fatal error repeating more than 100 times:

"LINK : fatal error LNK1181: cannot open input file 'delayimp.lib' [c:\UnrealEngine5_carla\Engine\Intermediate\ProjectFiles\UE5.vcxproj]
[image]

J160KU, 2024-07-30:
FATAL ERROR //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

**After fixing this, I stumbled into the next red fatal error** (which is repeating more than 100 times) **during compiling in the build process** (which is not interrupted, even by this fatal error): 

"LINK : fatal error LNK1181: cannot open input file 'delayimp.lib' [c:\UnrealEngine5_carla\Engine\Intermediate\ProjectFiles\UE5.vcxproj]
[image]

J160KU, 2024-07-30:
Hi @jeromepatel,

thanks for the quick reply. The right solution to my issue was to enable one of the MSVC VS C++ build tools packages. It was not the one you've mentioned. I can't tell you exactly which version it was yet, because I have enabled all of them now.

J160KU, 2024-07-30:
Now having solved fatal error LNK1181, the next errors named **LNK2019** are following:

[image]
[image]

zhaosiwen1949, 2024-08-01:
@J160KU I have the same error. With @jeromepatel 's help，I follow the steps:
1. Delete Build Folder;
2. install the MSVC 14.36.32532;
3. run /Setup.bat again;

[image]

Now I got the same error as above.

MarcelPiNacy-CVC, 2024-08-01:
Hi,

A workaround for this problem is to build CARLA with Visual Studio 2022 directly.
Make sure to delete the Build folder, perform a git pull, then open the CARLA folder with VS and select the Windows-Development CMake Preset:
[image]

Once that's done keep in mind that the new Build folder is Build/Windows-Development, so commands like `cmake --build * --target launch` should now target `Build/Windows-Development` instead of `Build`.

Kind regards,
Marcel

zhaosiwen1949, 2024-08-02:
@MarcelPiNacy-CVC Thanks for the help! I finnally make it successful!
Here's my steps:
1. First as @MarcelPiNacy-CVC said, use Visual Studio's Widows-Development Cmake Preset;
2. Run CMD.exe as Administrator. And use vcallvars.bat to set up the MSVC Tool's Version, like below:
`"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvarsall.bat" x64 10.0.20348.0 -vcvars_ver=14.36`
3. use `cmake --build Build/Windows-Development --target launch`
 Follow above steps, I finally make no error! Hope It's useful for other's problem~

closed 2024-08-02 by MarcelPiNacy-CVC

J160KU, 2024-08-06:
Hello,

where can I find the " "Visual Studio's Widows-Development Cmake Preset". It is not listed in my VS2022.

[image]

J160KU, 2024-08-06:
CMake Error //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

After proceeding with the CMake preset "x64-Debug" I get this new error message:
[image]

donghaiwang, 2025-10-17:
I've got the same error in Window 10 with ue4-dev branch.
```
D:\hutb\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/PixelReader.h(212): error C2065: “header_offset”: 未声明的标识符
```