carla-simulator/carla #7738: Weird behavior of traffic lights on intersections

dx6nxr, 2024-06-03:
CARLA version: 0.9.15
Platform/OS: Win11
Problem you have experienced: Traffic lights in opposite directions in intersections are not changed to green at the same time, but they kind of change to green in a "circle", instead of changing the traffic lights in opposing directions to green simultaneously. 
Steps to reproduce: traffic lights in intersections on any map

[image]
Example is from the map "Town05". Behavior on the different maps is the same.

I've been trying to find any info if somebody had the same issue, however i couldn't find anything issued about it on gh or any info in carla docs.

Best regards, 
Ivan

GoodarzMehr, 2024-06-04:
This is the default behavior.

"The traffic lights in the same group follow a cycle. The first one is set to green while the rest remain frozen in red. The active one spends a few seconds in green, yellow and red, so there is a period of time where all the lights are red. Then, the next traffic light starts its cycle, and the previous one is frozen with the rest."

[Source](https://carla.readthedocs.io/en/latest/core_actors/#traffic-signs-and-traffic-lights)

dx6nxr, 2024-06-04:
Hey @GoodarzMehr ,
thanks for the clarification. So, I guess there is no out-of-the-box way to change them to more "normal" color changing behavior? 

Best, Ivan

GoodarzMehr, 2024-06-04:
You're welcome. And to answer your question, not as far as I know, though you can write the code to get the [traffic lights](https://carla.readthedocs.io/en/latest/python_api/#carla.TrafficLight) at the intersection and set their states manually.

glopezdiest, 2024-06-07:
Following up on what @GoodarzMehr said, if both traffic lights were green, traffic would have to understand the concept of give way, which isn't a simple logic to design.

Closing this is at has been answered

closed 2024-06-07 by glopezdiest