carla-simulator/carla #7739: vehicle spawn segmentation fault wiht UE5

dvdhfnr, 2024-06-03:
Hi!

I followed [this doc](https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.5piein3604b7) to build the ue5-dev branch on Ubuntu 22.04. Building and packaging works fine. I am also able to spawn a camera actor and read images in python. However, whenever I try to spawn a vehicle actor, the server terminates with a segmenation fault. This is supposed to be working right now?

Thanks!

PeterChang1, 2024-06-05:
> Hi!
> 
> I followed [this doc](https://docs.google.com/document/d/1aWcoJr-9z_wE6S442UhzC32jUGmN1o42REE7C-4aGYA/edit#heading=h.5piein3604b7) to build the ue5-dev branch on Ubuntu 22.04. Building and packaging works fine. I am also able to spawn a camera actor and read images in python. However, whenever I try to spawn a vehicle actor, the server terminates with a segmenation fault:
> 
> ```
> bpl = world.get_blueprint_library()
> bps = [bp for bp in bpl.filter("vehicle.*") if int(bp.get_attribute("number_of_wheels")) == 4]
> spawn_points = world.get_map().get_spawn_points()
> actor = world.spawn_actor(bps[0], spawn_points[0]) # server terminates HERE
> ```
> 
> ->
> 
> ```
> 5.3.2-0+UE5 1009 0
> Disabling core dumps.
> Signal 11 caught.
> Malloc Size=262146 LargeMemoryPoolOffset=262162
> CommonUnixCrashHandler: Signal=11
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)
> ```
> 
> With lldb:
> 
> ```
> * thread #1, name = 'CarlaUnreal-Lin', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
>     frame #0: 0x0000000008e416d7 CarlaUnreal-Linux-Shipping`ACarlaWheeledVehicle::GetMaximumSteerAngle() const [inlined] FObjectPtr::operator!(this=0x0000000000000000) const at ObjectPtr.h:142:65
> ```
> 
> Is this supposed to be working right now?
> 
> Thanks!

I found that only vehicle.lincoln.mkz_2020 could be spawn successfully. Because other vehicles haven't add the wheel Setup. Spawn your vehicle filter by "vehicle.lincoln.mkz_2020".

Blyron, 2024-06-05:
Hey if you get latest content by updating content repo, and you update code repo, you will be able to have 6 cars up and running

dvdhfnr, 2024-06-05:
Thanks. This works now.

closed 2024-06-05 by dvdhfnr

PeterChang1, 2024-06-06:
> Hey if you get latest content by updating content repo, and you update code repo, you will be able to have 6 cars up and running

May I know which 6 cars can work? There are 41 Cars in Vehicle.*

Blyron, 2024-06-06:
Look at vehicles factory blueprint
Or do a Python  blueprint_list = get_actor_blueprints(self.world, vehicle.* , self._actor_generation)  in latest changes or ue5-dev