carla-simulator/carla #7756: pygame window not working properly.

jayd972, 2024-06-05:
Carla version: 0.9.15
System: Windows 11, 4GB Nvidia RTX 3050Ti GPU, Ryzen 6800H CPU, 16GB DDR5 RAM

I installed Carla and Python API, it seems to work well and I can run the generate_traffic and dynamic_weather examples.

When I try running the manual_control example, a pygame window is opened with a vehicle. After that both UE and pygame freezes and not responding.

Any help would be appreciated.

Blyron, 2024-06-05:
Your GPU has too low VRAM so it won't be sufficient to run a pygame mode. You still can run no rendering mode. or agents, but not launch pygame

jayd972, 2024-06-05:
will it help if I degrade the Carla version?

Blyron, 2024-06-06:
You will have to go to a very old one 0.8.X which is not supported anymore

closed 2024-06-12 by Blyron