carla-simulator/carla #7759: Issue while Packaging the Custom Map Created Using Digital Twin Tool

NidhinPPramod, 2024-06-06:


Blyron, 2024-06-06:
you have to use make package command, with the DefaultGame.ini edited to use your map. Remember including /CustomMaps/YourMap/Opendrive/ folder as it is need to launch the maps.

NidhinPPramod, 2024-06-06:
> you have to use make package command, with the DefaultGame.ini edited to use your map. Remember including /CustomMaps/YourMap/Opendrive/ folder as it is need to launch the maps.

Should I add the opendrive path in the Additional non asset directories

Blyron, 2024-06-06:
Change default map in DefaultGameMode.ini and add your maps to cook, then do a make package

NidhinPPramod, 2024-06-06:
> Change default map in DefaultGameMode.ini and add your maps to cook, then do a make package
This is my packaging error looks like:
[image]
Packaging and Game Mode settings:
[image]
[image]

Blyron, 2024-06-06:
Look how the ini file is set... 
[image]

Blyron, 2024-06-06:
Something on your map is broken we need more info about the packaging

NidhinPPramod, 2024-06-06:
> Look how the ini file is set... [...]

**This is my .ini file**

[/Script/EngineSettings.GeneralProjectSettings]
ProjectID=675BF8694238308FA9368292CC440350
ProjectName=CARLA UE4
CompanyName=CVC
CopyrightNotice="Copyright (c) 2017 Computer Vision Center (CVC) at the Universitat Autonoma de Barcelona (UAB). This work is licensed under the terms of the MIT license. For a copy, see <https://opensource.org/licenses/MIT>."
ProjectVersion=0.9.15

[/Script/Carla.CarlaSettings]
+LowRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad.CheapRoad"',MaterialSlotName="Tileroad_Road",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+LowRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb.CheapSideWalkCurb"',MaterialSlotName="TileRoad_Curb",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+LowRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00.CheapSideWalk_00"',MaterialSlotName="Tileroad_Sidewalk",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+LowRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking.CheapLaneMarking"',MaterialSlotName="TileRoad_LaneMarkingSolid",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
LowLightFadeDistance=1000.000000
LowStaticMeshMaxDrawDistance=10000.000000
LowRoadPieceMeshMaxDrawDistance=15000.000000
+EpicRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2.WetPavement_Complex_Road_N2"',MaterialSlotName="Tileroad_Road",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+EpicRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete.WetPavement_Complex_Concrete"',MaterialSlotName="TileRoad_Curb",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+EpicRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4.WetPavement_SidewalkN4"',MaterialSlotName="Tileroad_Sidewalk",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))
+EpicRoadMaterials=(MaterialInterface=MaterialInstanceConstant'"/Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking.Lanemarking"',MaterialSlotName="TileRoad_LaneMarkingSolid",ImportedMaterialSlotName="",UVChannelData=(bInitialized=False,bOverrideDensities=False,LocalUVDensities[0]=0.000000,LocalUVDensities[1]=0.000000,LocalUVDensities[2]=0.000000,LocalUVDensities[3]=0.000000))

[/Script/UnrealEd.ProjectPackagingSettings]
Build=IfProjectHasCode
BuildConfiguration=PPBC_Development
BuildTarget=
StagingDirectory=(Path="")
FullRebuild=False
ForDistribution=False
IncludeDebugFiles=True
BlueprintNativizationMethod=Disabled
bIncludeNativizedAssetsInProjectGeneration=False
bExcludeMonolithicEngineHeadersInNativizedCode=False
UsePakFile=True
bUseIoStore=False
bGenerateChunks=False
bGenerateNoChunks=False
bChunkHardReferencesOnly=False
bForceOneChunkPerFile=False
MaxChunkSize=0
bBuildHttpChunkInstallData=False
HttpChunkInstallDataDirectory=(Path="")
PakFileCompressionFormats=
PakFileAdditionalCompressionOptions=
HttpChunkInstallDataVersion=
IncludePrerequisites=True
IncludeAppLocalPrerequisites=False
bShareMaterialShaderCode=False
bDeterministicShaderCodeOrder=False
bSharedMaterialNativeLibraries=False
ApplocalPrerequisitesDirectory=(Path="")
IncludeCrashReporter=False
InternationalizationPreset=English
-CulturesToStage=en
+CulturesToStage=en
LocalizationTargetCatchAllChunkId=0
bCookAll=False
bCookMapsOnly=False
bCompressed=False
bSkipEditorContent=False
bSkipMovies=False
-IniKeyBlacklist=KeyStorePassword
-IniKeyBlacklist=KeyPassword
-IniKeyBlacklist=rsa.privateexp
-IniKeyBlacklist=rsa.modulus
-IniKeyBlacklist=rsa.publicexp
-IniKeyBlacklist=aes.key
-IniKeyBlacklist=SigningPublicExponent
-IniKeyBlacklist=SigningModulus
-IniKeyBlacklist=SigningPrivateExponent
-IniKeyBlacklist=EncryptionKey
-IniKeyBlacklist=IniKeyBlacklist
-IniKeyBlacklist=IniSectionBlacklist
+IniKeyBlacklist=KeyStorePassword
+IniKeyBlacklist=KeyPassword
+IniKeyBlacklist=rsa.privateexp
+IniKeyBlacklist=rsa.modulus
+IniKeyBlacklist=rsa.publicexp
+IniKeyBlacklist=aes.key
[... cut ...]
+MapsToCook=(FilePath="/Game/CustomMaps/Hoodi2Btp/Hoodi2Btp")
+MapsToCook=(FilePath="/Game/CustomMaps/Hoodi2Btp/Hoodi2Btp/OpenDrive")
+DirectoriesToAlwaysCook=(Path="/Game/Carla/Static/GenericMaterials/Licenseplates/Textures")
+DirectoriesToAlwaysCook=(Path="/Game/Carla/Static/Car/4Wheeled/ParkedVehicles")
+DirectoriesToAlwaysStageAsUFS=(Path="Carla/Maps/OpenDrive")
+DirectoriesToAlwaysStageAsUFS=(Path="Carla/Maps/Nav")
+DirectoriesToAlwaysStageAsUFS=(Path="Carla/Maps/TM")
+DirectoriesToAlwaysStageAsUFS=(Path="Carla/Config")

Blyron, 2024-06-06:
+MapsToCook=(FilePath="/Game/CustomMaps/Hoodi2Btp/Hoodi2Btp/OpenDrive")
this incorrect, this should be a DirectoriesToAlwaysStageAsUFS
+DirectoriesToAlwaysStageAsUFS=("CustomMaps/Hoodi2Btp/Hoodi2Btp/OpenDrive")

NidhinPPramod, 2024-06-06:
> Something on your map is broken we need more info about the packaging
**cook file**
[Cook-2024.06.06-15.33.48.txt](https://github.com/user-attachments/files/15613361/Cook-2024.06.06-15.33.48.txt)
**log file**
[Log.txt](https://github.com/user-attachments/files/15613362/Log.txt)

Blyron, 2024-06-06:
This is an issue related with Unreal itself, i won'tbe able to help you. Is related with the seettings of the generated mesh. 
Sorry for this. Look for help in Unreal forum

Blyron, 2024-06-06:
The issue is the mesh itself, one of the generated meshes seems to have an issue. We are still working on DT tool as it was a very first version. We are not able to help right now. But now we are aware of this issue we will look into that in next revisions of DT Tool.

NidhinPPramod, 2024-06-06:
> The issue is the mesh itself, one of the generated meshes seems to have an issue. We are still working on DT tool as it was a very first version. We are not able to help right now. But now we are aware of this issue we will look into that in next revisions of DT Tool.

Is there any alternate solution I can use ,to run this custom map as a standalone server

Blyron, 2024-06-06:
Not right now. Sorry. The workaround it make it run in editor mode.

NidhinPPramod, 2024-06-06:
> Not right now. Sorry. The workaround it make it run in editor mode.

Is there any option like i can run this map server in my pc and the client from other pc can use it..like using IP address or something

Blyron, 2024-06-06:
Which is your GPU?

NidhinPPramod, 2024-06-06:
> Which is your GPU? You cannot connect from another computer.

Nvidia RTX A5000
Is there any way like i can use this custom map in another pc?

NidhinPPramod, 2024-06-06:
> > Which is your GPU? You cannot connect from another computer.
> 
> Nvidia RTX A5000 Is there any way like i can use this custom map in another pc?Also, when can i expect the fix in DT Tool

Blyron, 2024-06-06:
> > Which is your GPU? You cannot connect from another computer.
> 
> Nvidia RTX A5000 Is there any way like i can use this custom map in another pc?Also, when can i expect the fix in the DT Tool

You can use the map in editor mode. What you cannot do is create a package with it. If you want to, you have to fix the conflict mesh on your own. ( I do not know how to fix it, look at Unreal forum):

When can you expect the fix for DT Tool? 
Do not know. Not time determined.

closed 2024-06-06 by Blyron