carla-simulator/carla #7767: Recorder / Replayer improvements

glopezdiest, 2024-06-07:
Here are several improvements for the recorder / replayer functionalities
1) When running `client.start_recorder`, the function will fail if the provided path is invalid, but it doesn't provide any warning or error
2) Same goes for the `client.show_recorder_file_info` where instead of returning a warning / error, it creates a file with the message "Provided path is invalid"
3) Add a function similar to `show_recorder_file_info` that parses the replayer information to a json or similar format
4) Create an easy way to replay a file from the point of view of the ego vehicle, either by adding a `follow_ego`,, by making an easy way to get it's id, or any other possibility
5) The replayer currently teleports vehicles as away to ensure the replayed behavior matches the recorded one, losing information about the control, velocity and acceleration of the actors. Find a way to keep that information in the replayer. A possible test would be to see what happens if you set the control, velocity and acceleration of the actors, but still keep the teleporting them

PatrickPromitzer, 2024-06-07:
Hi,
I know it is not supposed to add a command on this kind of issue, but would it be possible to add attached sensor transformation to the recording?
I currently don't use the replay function, but it could be a "good to have" for some people.

For example: 
actor_1
  attached_sensors: 
    - sensor.lidar.ray_cast: x=0.0, y=0.0, z=0.0, roll=0.0, pitch=0.0, yaw=0.0

I believe if someone wants to record the sensor data at a later date, it should be enough information to do so.

glopezdiest, 2024-06-21:
> I know it is not supposed to add a command on this kind of issue

I disagree, feel free to comment in these. Any user feedback is important here, as it shows what the community is interested in, what could benefit them, and what the overall consensus on the feature is.

As for your specific request, the replay_file already allows the replaying of the sensors used during the recording, isn't that already what you want?

PatrickPromitzer, 2024-06-21:
My bad, I thought the sensors were excluded from the actor list.

closed 2026-11-12 by Madecu