carla-simulator/carla #7770: Error Building PythonAPI with OSM2ODR on Windows

CodeManageSystem, 2024-06-10:
### GitHub Issue Description

**Title**: Error Building PythonAPI with OSM2ODR on Windows

**Description**:
I encountered issues while building the PythonAPI for Carla, specifically related to the OSM2ODR component on a Windows environment. Below is the detailed log and error messages observed during the build process.

**Steps to Reproduce**:
1. Follow the standard build process for Carla on Windows.
2. Attempt to build the PythonAPI using the `make PythonAPI` command.

**Observed Behavior**:
The build process fails with the following errors:

```
-[BuildOSM2ODR]: [Batch params]: --generator "" --build --all
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  196M    0  196M    0     0   943k      0 --:--:--  0:03:33 --:--:-- 1571k
tar: This does not look like a tar archive
tar: Skipping to next header
tar: Exiting with failure status due to previous errors
系统找不到指定的文件。
CMake Error: The source directory "E:/Carla/carla/Build/osm2odr-source"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
系统找不到指定的文件。
系统找不到指定的文件。
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "E:\Carla\carla\PythonAPI\carla\dependencies\"

-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.9
creating build\lib.win-amd64-3.9\carla
copying source\carla\command.py -> build\lib.win-amd64-3.9\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.9\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.9
creating build\temp.win-amd64-3.9\Release
creating build\temp.win-amd64-3.9\Release\source
creating build\temp.win-amd64-3.9\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\Jason\AppData\Local\Programs\Python\Python39\include -IC:\Users\Jason\AppData\Local\Programs\Python\Python39\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\cppwinrt -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\cppwinrt /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失
E:\Carla\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: 无法打开包括文件: “OSM2ODR.h”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
```

**Additional Context**:
- The log suggests an issue with the tar extraction, indicating that the tar file does not look like a valid archive.
- The directory `E:/Carla/carla/Build/osm2odr-source` seems to be missing, resulting in a failure to load the cache.
- During the compilation step, there is a fatal error related to missing the `OSM2ODR.h` file.
- The build process appears to skip files and headers leading to a build failure.

**Environment**:
- Operating System: Windows
- Visual Studio Version: 2019 Community + sdk10.0.20348.0
- Python Version: 3.9.6

Thank you for your support!

muety, 2024-06-11:
#6898 might solve your problem.

closed 2024-10-19 by Blyron

qshjd-123, 2024-11-22:
Have you resolved it? I've encountered the same error.

aFewThings, 2025-03-27:
I faced same problem, and I solved it by directly downloading zip file from here (https://github.com/carla-simulator/sumo/archive/%CURRENT_OSM2ODR_COMMIT%.zip). 

In BuildOSM2ODR.bat, `%CURRENT_OSM2ODR_COMMIT%` is set to `1835e1e9538d0778971acc8b19b111834aae7261`.

So, you need to download ```https://github.com/carla-simulator/sumo/archive/1835e1e9538d0778971acc8b19b111834aae7261.zip```, extract the folder, rename extracted folder to `osm2odr-source`, then place the folder into `Build` folder.

These steps are identical to the commands in BuildOSM2ODR.bat.

```
rem Build OSM2ODR
if %BUILD_OSM2ODR% == true (
    cd "%INSTALLATION_DIR%"
    if not exist "%OSM2ODR_SOURCE_PATH%" (
        curl --retry 5 --retry-max-time 120 -L -o OSM2ODR.zip https://github.com/carla-simulator/sumo/archive/%CURRENT_OSM2ODR_COMMIT%.zip
        tar -xf OSM2ODR.zip
        del OSM2ODR.zip
        ren sumo-%CURRENT_OSM2ODR_COMMIT% osm2odr-source
    )

```

I'm not sure exactly what causes the problem, but it seems to be caused by the `tar -xf OSM2ODR.zip` command.