carla-simulator/carla #7772: My Code is launching pygame window when i run a map with single tile, But im getting segmentation fault when i try to use a Map having multiple tiles and aslo Server gets crashed.

Chethan2003, 2024-06-11:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04
Problem you have experienced:

LoginId:a2d768913f5dd54840cc723464199297-000003e9
EpicAccountId:
 
Caught signal 11 Segmentation fault
 
libUE4Editor-Carla.so!carla::streaming::detail::token_type::operator carla::streaming::Token() const(+0x3)
libUE4Editor-Carla.so!ASensor::EndPlay(EEndPlayReason::Type) [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/Sensor.cpp:120]
libUE4Editor-Carla.so!ASceneCaptureCamera::EndPlay(EEndPlayReason::Type) [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureCamera.cpp:46]
libUE4Editor-Engine.so!AActor::Destroyed() [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/Actor.cpp:2332]
libUE4Editor-Engine.so!UWorld::DestroyActor(AActor*, bool, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/LevelActor.cpp:708]
libUE4Editor-Engine.so!AActor::Destroy(bool, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/Actor.cpp:4056]
libUE4Editor-Carla.so!FActorRegistry::PutActorToSleep(unsigned int, UCarlaEpisode*) [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorRegistry.cpp:198]
libUE4Editor-Carla.so!FActorRegistry::PutActorToSleep(unsigned int, UCarlaEpisode*) [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorRegistry.cpp:201]
libUE4Editor-Carla.so!ALargeMapManager::ConvertActiveToDormantActors() [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp:817]
libUE4Editor-Carla.so!ALargeMapManager::Tick(float) [/home/akshat/Nidhin_wrkspace/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/LargeMapManager.cpp:361]
libUE4Editor-Engine.so!AActor::TickActor(float, ELevelTick, FActorTickFunction&) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/Actor.cpp:1091]
libUE4Editor-Engine.so!FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/Actor.cpp:173]
libUE4Editor-Engine.so!FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:284]
libUE4Editor-Engine.so!TGraphTask<FTickFunctionTask>::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:886]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:709]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:600]
libUE4Editor-Core.so!FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr<FGraphEvent>, TInlineAllocator<4u, TSizedDefaultAllocator<32> > > const&, ENamedThreads::Type) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:1524]
libUE4Editor-Engine.so!FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:563]
libUE4Editor-Engine.so!FTickTaskManager::RunTickGroup(ETickingGroup, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1577]
libUE4Editor-Engine.so!UWorld::Tick(ELevelTick, float) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1457]
libUE4Editor-Engine.so!UGameEngine::Tick(float, bool) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Engine/Private/GameEngine.cpp:1791]
UE4Editor!FEngineLoop::Tick() [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4830]
UE4Editor!GuardedMain(char16_t const*) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/akshat/UnrealEngine-carla/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xf2)
UE4Editor!_start()

has context menu

glopezdiest, 2024-06-21:
Closing the issue, please create a new one with the correct format.

Titles shouldn't be the description of the issue, but a short phrase to indicate the issue. Use the "Add a description" box to describe the actual issue, as well as detailed information about how to reproduce it, and any other information that might be relevant

closed 2024-06-21 by glopezdiest

amckendr, 2026-08-16:
We hit this crash class in production on CARLA **0.9.16** (packaged Linux release, UE4.26 Shipping, Vulkan, `-RenderOffScreen`, Ubuntu 24.04, RTX GPU) and can now reproduce it **deterministically, in seconds**, with full symbolized backtraces (cores captured via apport, symbolized against the `CarlaUE4-Linux-Shipping.debug` that ships beside the binary). Posting here because this issue's original stack is the same crash and it was closed for formatting only.

## Mechanism

On a Large Map (`Town12` here), `ALargeMapManager`'s dormancy sweep crashes the server whenever it puts **any non-hero actor that is (or carries) a sensor** to sleep:

```
ALargeMapManager::Tick
  → ALargeMapManager::ConvertActiveToDormantActors   (LargeMapManager.cpp:870)
  → UCarlaEpisode::PutActorToSleep
  → FActorRegistry::PutActorToSleep                  (ActorRegistry.cpp:198, recurses into attached children at :201)
  → FCarlaActor::PutActorToSleep                     (CarlaActor.cpp:161)
  → AActor::Destroy → AActor::Destroyed → AActor::RouteEndPlay
  → ASensor::EndPlay                                 (Sensor.cpp:129)
  → FDataStreamTmpl<…MultiStreamState>::GetToken     (DataStream.h:55)
  → carla::streaming::detail::token_type::operator Token()   ← SIGSEGV
```

Both sensor shapes die on this path, confirmed by separate cores:

- an **attached** sensor (IMU/camera on a vehicle): the parent vehicle is swept and `FActorRegistry::PutActorToSleep` recurses into the child sensor;
- a **parentless** sensor (a free-standing `sensor.camera.rgb`): swept directly, via `ASceneCaptureCamera::EndPlay` (SceneCaptureSensor.cpp:617).

We also captured the mid-tick sibling of this crash on a live world: `AInertialMeasurementUnit::PostPhysTick` → `AInertialMeasurementUnit::GetActorAngularVelocityInRadians` → `AActor::GetRootComponent(this=0x0)` (InertialMeasurementUnit.cpp:58) — an IMU ticking after its parent was already gone, i.e. the same sleep/destroy-vs-sensor-lifecycle race seen from the other side.

## Controlled result (one variable per cell)

| spawn | map | `role_name` | outcome |
|---|---|---|---|
| vehicle + attached IMU + RGB cam, sync 0.02 | Town12 | *(none)* | **server SIGSEGV within seconds — 3/3 runs** |
| identical | Town12 | `hero` | survived 300 s / 15,358 ticks, clean |
| identical | Town10HD_Opt | *(none)* | survived 300 s / 43,979 ticks, clean |
| parentless RGB cam far from spectator, async | Town12 | *(none)* | **server SIGSEGV in under a second** (swept on the next tick) |
| parentless RGB cam, same position, async | Town12 | `hero` | survived 150 s / 12,481 frames, clean destroy |

Hero-tagged actors are exempt from the sweep and single-tile maps have no `LargeMapManager`, so flipping either variable takes the crash rate from "seconds" to zero. In an **async** world the sweep runs every engine tick, so a non-hero sensor beyond the active distance kills the server almost instantly; under synchronous ticking it takes a few seconds of ticks.

## Minimal repro (0.9.16 wheel, default Town12 server)

```python
import carla
client = carla.Client("127.0.0.1", 2000); client.set_timeout(20.0)
world = client.get_world()   # server on Town12 (any Large Map)

world.get_spectator().set_transform(
    carla.Transform(carla.Location(x=0, y=0, z=50)))
bps = world.get_blueprint_library()
sp = world.get_map().get_spawn_points()
far = max(sp, key=lambda t: t.location.x**2 + t.location.y**2)

bp = bps.find("sensor.camera.rgb")
# bp.set_attribute("role_name", "hero")   # <- uncomment and it survives
cam = world.spawn_actor(bp, carla.Transform(
    carla.Location(x=far.location.x, y=far.location.y, z=300.0),
    carla.Rotation(pitch=-90.0)))
cam.listen(lambda d: None)               # server is usually dead already
import time; time.sleep(10)              # SIGSEGV within one sweep tick
```

The attached-sensor variant is the same with a vehicle + `attach_to=` IMU/camera, `synchronous_mode=True`, `fixed_delta_seconds=0.02`, destroying the **vehicle first** and continuing to tick.

## Why this bites real users

Any client that dies without destroying its sensors (crash, SIGKILL, an uncaught exception in a libcarla callback thread) leaves an orphan; on a Large Map the next dormancy sweep then kills the **server**. That converts every client-side failure into a server crash a few seconds later, which looks like random server instability until the cores are symbolized. It is also, we believe, why this class is reported almost exclusively on multi-tile maps (as in this issue and the original report's stack, which runs through the same `ConvertActiveToDormantActors` frame).

## Workarounds that we verified

- Tag sensors (or their parent vehicle) `role_name=hero` — exempt from the sweep.
- Keep sensors attached to hero vehicles, or within `actor_active_distance` of a hero/spectator anchor.
- In cleanup paths, `stop()` + `destroy()` sensors **before** destroying their parent, and sweep orphaned sensors before working on a Large Map world.

Happy to share the full gdb `thread apply all bt` outputs from the cores (attached-sensor case, parentless case, and the `PostPhysTick` mid-tick case) if useful — the fault address in `token_type::operator Token()` is identical across all sweep-path cores.