carla-simulator/carla #7782: Unable to build CARLA UE5.3 on Windows - Error generating pnglibconf.out

WilliamFlinchbaugh, 2024-06-12:
I've been try to build the UE5.3 version of CARLA for a bit now and I can't seem to solve this problem. 

I'm able to install all dependencies and configure fine, but when I run `cmake --build Build`, it fails immediately. Looks like there's some problem with libpng.

I'm on Windows 10 and I simply cloned the dev-ue5 branch and ran Setup.bat

Here's the log:

```
-- CARLA: Configuring file E:/CarlaUE5/LibCarla/source/carla/Version.h
-- CARLA: Configuring file E:/CarlaUE5/PythonAPI/carla/pyproject.toml
-- CARLA: Configuring file E:/CarlaUE5/PythonAPI/carla/CMakeLists.txt
-- CARLA: (option) ENABLE_DIRECTORY_CLEAN : ON
-- CARLA: CARLA Unreal Content - FOUND
-- Configuring done (6.8s)
-- Generating done (1.7s)
-- Build files have been written to: E:/CarlaUE5/Build
Buiding CARLA...
[1/284] Generating pnglibconf.out
FAILED: _deps/libpng-build/pnglibconf.out E:/CarlaUE5/Build/_deps/libpng-build/pnglibconf.out
C:\Windows\system32\cmd.exe /C "cd /D E:\CarlaUE5\Build\_deps\libpng-build && "C:\Program Files\CMake\bin\cmake.exe" -DINPUT=E:/CarlaUE5/Build/_deps/libpng-build/pnglibconf.c -DOUTPUT=E:/CarlaUE5/Build/_deps/libpng-build/pnglibconf.out -P E:/CarlaUE5/Build/_deps/libpng-build/scripts/genout.cmake"
Microsoft (R) C/C++ Optimizing Compiler Version 19.40.33811 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

pnglibconf.c
E:/CarlaUE5/Build/_deps/zlib-src\zlib.h(34): fatal error C1083: Cannot open include file: 'zconf.h': No such file or directory
CMake Error at scripts/genout.cmake:78 (message):
  Failed to generate E:/CarlaUE5/Build/_deps/libpng-build/pnglibconf.out.tf1

[18/284] Building CXX object _deps\boost-build\libs\serial...CMakeFiles\boost_serialization.dir\src\xml_grammar.cpp.obj
ninja: build stopped: subcommand failed.
```

Blyron, 2024-06-13:
Try to delete build folder inside carlaue5 and then run setup.bat in a x64 vs 2022 terminal as administrator

WilliamFlinchbaugh, 2024-06-16:
> Try to delete build folder inside carlaue5 and then run setup.bat in a x64 vs 2022 terminal as administrator

I tried this, but I got the same error.

I've got some other weird results too. So I tried to reproduce this issue on a different machine running Windows 11, but I didn't run into this issue. I ran into a different issue, but that seemed to be a regular compiler error. 

Regardless, I think that the Windows build seems to be pretty wonky ATM. I was able to run a dual-boot with Ubuntu 22.04 and it built first try with no issues.

closed 2024-06-16 by WilliamFlinchbaugh

J160KU, 2024-07-17:
Hello @WilliamFlinchbaugh,

maybe it works when running the Setup.bat in the Git terminal, which is included in Git -> https://git-scm.com/downloads

jeromepatel, 2024-07-29:
I encountered both the issues mentioned here, encountered by @WilliamFlinchbaugh and @J160KU , commenting for future reference here the solutions:

**Solution for Issue with pnglibconf.out:** 
Reason for this error is somewhere in `CarlaSetup.bat` script command, the file `zconf.h` is being renamed to `zconf.h.included` (of course there must be a reason to add that line previously but it has no significance now). 
The solution is simply rename the file Rename the `CarlaUE5\Build\_deps\zlib-src\zconf.h.included` to `zconf.h` and then run this command next 
```
cmake --build Build
```
 which should hopefully give a successful build. and then Run Python API build and launch command. 
For me, next error occurred with `cmake --build Build --target launch` command. If your launch is successful then congrats but it might be possible that this next issue persists. 

**Solution to AsyncDataStreamImpl.h: Error c2510:...:**
With the help of [this reddit post](https://www.reddit.com/r/unrealengine/comments/1axxe63/can_anyone_help_me_with_this_unreal_5_error/) I resolved this issue. The fix was to go into VS Installer and enable the option under > Modify > Individual Components > MSVC v143 - VS 2022 C++ x64/x86 build tools (v14.36-17.6).

But only this will not be enough, SO, Now delete “Build” folder and  open from x64 native tools VS 2022 terminal with administrator, navigate to Carla folder and run the “CarlaSetup.bat” which will probably give the first error again Since the script rewrites the filename again, So simply follow, the first solution again and this time launch command shouldn't be problem :)

I kept the second error solution separate since both issues are not linked and it might be possible that for someone there's no need to install MSVC v143 build tools.

J160KU, 2024-07-30:
@jeromepatel **Thank you for the advice!**

Modifying the Visual Studio Installer solved my error messages. I will continue my topic in this issue: **https://github.com/carla-simulator/carla/issues/7737**