carla-simulator/carla #7785: Segfault when cooking OSM map

AreopagX, 2024-06-14:
Hi all,
I'm currently trying to import a simple OSM map using the digital twin feature. However, when I try to export this asset using `make package ARGS="--package=MyMap"` then the cooking process crashes with a segmentation fault. Can you help me get this running?

Detailed steps:

- checkout origin/dev using git
- run `make osmrenderer`
- run `make launch`
- find "UW_OnRoadMainWidget" in "CarlaTools Content" within the Content Browser of the UE4 Editor and run it
- fill the fields "OSM URL" with "https://overpass-api.de/api/map?bbox=8.56931,50.15577,8.57543,50.16098" and "File Name" with "MyMap"
- click "Generate" and when things have finished loading, click "Save Map"
- now, there should be a folder "Unreal/CarlaUE4/CustomMaps/MyMap"
- create a file "MyMap.Package.json" in "Unreal/CarlaUE4/Content/Carla/Config" with the following contents:
``` json

{
  "maps": [
    {
        "path": "/Game/CustomMaps/MyMap",
        "name": "MyMap",
        "use_carla_materials": true
      }
  ],
  "props": []
}
```
- then run `make package ARGS="--package=MyMap"`

In my case this script encounters a segmentation fault when it starts cooking the assets. You can find the crash logs below:

[CarlaUE4.log](https://github.com/user-attachments/files/15838210/CarlaUE4.log)
[CrashReportClient.ini.txt](https://github.com/user-attachments/files/15838221/CrashReportClient.ini.txt)
[Diagnostics.txt](https://github.com/user-attachments/files/15838222/Diagnostics.txt)
[CrashContext.runtime-xml.txt](https://github.com/user-attachments/files/15838225/CrashContext.runtime-xml.txt)

This issue must be related to the digital twin tool / the map specifically since packaging "Town06_Opt" using the same command works.

Blyron, 2024-06-14:
This map seems to have meshes with issues. We still making progress on Digital Twins tool

AreopagX, 2024-06-17:
@Blyron  So far I couldn't find a map that can be cooked. Can you share links to OSM maps, which work with the digital twins tool? Thanks in advance.

Blyron, 2024-06-17:
Not really. Haven't found, we will be working on it

Blyron, 2025-08-25:
This should be fixed now