carla-simulator/carla #7788: Unable to git clone CarlaUnreal

xiangyifeng, 2024-06-16:
Because of this https://forums.unrealengine.com/t/maintenance-on-the-epic-games-github-organization-june-2024/1891135,  I can’t access https://github.com/CarlaUnreal/UnrealEngine.git/.Dose anyone have a way to solve this problem.

WilliamFlinchbaugh, 2024-06-16:
This happens because the repository is private, and only those in the Epic Games GitHub Org can access it. You'll need to link your GitHub account with Epic Games, then join the Epic Games Org, and then you'll need to clone it using a personal access key:

1. Follow these instructions to link your account and join the Epic Games Organization: https://www.unrealengine.com/en-US/ue-on-github
2. Follow these instructions to setup your personal access token: https://github.com/carla-simulator/carla/issues/4543#issuecomment-2017157107
3. Wherever you are running your `git clone https://github.com/CarlaUnreal/UnrealEngine.git`, you'll have to switch it to `git clone https://oauth2:ghp_12345abcdefg@github.com/CarlaUnreal/UnrealEngine.git` where "ghp_12345abcdefg" is your personal access key.

rmrtcla, 2024-06-17:
> This happens because the repository is private, and only those in the Epic Games GitHub Org can access it. You'll need to link your GitHub account with Epic Games, then join the Epic Games Org, and [...]

@WilliamFlinchbaugh , thanks for your help. This is exactly what I did (and I guess also what @xiangyifeng  did). However, since June 10th, there's this change in how Epic organizes its Github Unreal Org, please see here: https://forums.unrealengine.com/t/maintenance-on-the-epic-games-github-organization-june-2024/1891135

Since then, it's not possible for new sign-ups to get access to the original Org, only to a mirror -- and unfortunately, the original CarlaUnreal fork was made from the original Org, not the new mirror. 

The same problem is discussed in here:  https://github.com/carla-simulator/carla/issues/7775

Blyron, 2024-06-28:
[image]

closed 2024-06-28 by Blyron