carla-simulator/carla #7790: CARLA is deployed on a remote server, with local display (server on remote, client on local).

ccconquer, 2024-06-17:
When deployed on a remote server, how should the following errors be resolved? When displaying the pygame window locally, the operation is very slow. How can this be improved?
[image]

GoodarzMehr, 2024-06-18:
These are normal, I think they just mean CARLA couldn't find a sound output on the server.

ccconquer, 2024-06-21:
> These are normal, I think they just mean CARLA couldn't find a sound output on the server.

Thanks! What does 'Disabling core dumps' mean? Does it matter?

GoodarzMehr, 2024-06-21:
AFAIK the Linux kernel creates a core dump (a snapshot of the program in system memory) at the moment a program crashes; this just disables it because CARLA is a rather memory-heavy program and creating a core dump would waste disk space and system resources.