carla-simulator/carla #7791: When using a remote headless server to host the Carla server and running the client locally, an issue arises.

ccconquer, 2024-06-17:
When running manual_control.py, the pygame window displays correctly on the local machine. However, when running programs like generate_traffic.py that do not involve pygame display, only text output appears locally with no image display. How can I achieve image display locally in such cases? Do you have any suggestions or solutions?
[image]

PatrickPromitzer, 2024-06-20:
You could run both scripts.
1.) start carla server
2.) start generate_traffic.py
3.) start manual_control.py

You are not limited to only one client connection.

glopezdiest, 2024-06-21:
By default, the spectator is the only way to have visual feedback of the simulation. The manual control is just a special client that transforms the sensor data into image for you to see. The solution proposed by @PatrickPromitzer is the best and most common one used

closed 2024-06-21 by glopezdiest