carla-simulator/carla #7793: Can a headless remote server serve as both the CARLA server and the CARLA client?

ccconquer, 2024-06-17:
Can a headless remote server serve as both the CARLA server and the CARLA client, effectively hosting the entire CARLA setup on the remote headless server?

ccconquer, 2024-06-17:
Running ./CarlaUE4.sh -carla-rpc-port=2000 -RenderOffScreen -graphicsadapter=1 -quality-level=Low on a headless server can start the server. When the client runs python3 generate_traffic.py --host 10.80.42.158 -p 2000, it produces results but no image is displayed. Are there any methods or suggestions to display the running images/videos, similar to remote streaming to some software?
[image]
[image]
[image]

PatrickPromitzer, 2024-06-20:
I guess you connect to the server with shh.

if you use Linux, try using 
```
ssh -X user@IP
```
The "-X" will forward X11 to the client.
I have an different command saved, but I forgot the behavior of the "-C" flag
```
ssh –Y –C user@IP
```

You need to set the forwarding permission on the server, or it will give an error.
```
xhost +
``` 

on windows, you need to install a software to emulate the X11 server.
[VcXsrv Windows X Server](https://sourceforge.net/projects/vcxsrv/)

ccconquer, 2024-06-21:
Thank you so much!! I'll try it.

Aperiss, 2024-06-21:
Thanks @PatrickPromitzer for the answer, we are closing this issue, but feel free to open if you run into any other issues.

closed 2024-06-21 by Aperiss