carla-simulator/carla #7795: How does CARLA represent occlusion？

ccconquer, 2024-06-17:
How does CARLA represent occlusion and the resulting blind spots (from the perspective of the ego vehicle, such as the blue car in the image), and how can it correctly obtain environmental information from the ego vehicle's perspective (e.g., which cars are occluded at certain times and thus not visible to the ego vehicle, meaning it cannot receive information about them)?
[image]
[image]

GoodarzMehr, 2024-06-18:
AFAIK it doesn't. You have to use the Python API to determine which cars are occluded.

ccconquer, 2024-06-21:
Thanks a lot! I will try to use the Python API to represent it.

Aperiss, 2024-06-21:
We are closing this issue, feel free to reopen if you run into any issues.