carla-simulator/carla #7803: set_autopilot() error in carla0.9.15

ccconquer, 2024-06-18:
I searched for information online, and some people suggested that the default port 8000 might be occupied. However, I tried many different ports and still encountered the error. What could be causing the error 'RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error'?
[image]

GoodarzMehr, 2024-06-18:
Are you creating the traffic manager and setting it to use port 2002 (which you are using for autopilot)? Are you using the synchronous mode?

You should try to provide your code, or the piece that's causing the error. Otherwise the answers you get here may not be helpful because nobody knows what you're doing.

ccconquer, 2024-06-21:
Thanks!! No, i didn't use synchronous mode and traffic manager. Do set_autopilot() only work in synchronous mode and must have a traffic manager?

GoodarzMehr, 2024-06-21:
Yes, you need to create a Traffic Manager so it can control vehicles autonomously. You can read more about the Traffic Manager [here](https://carla.readthedocs.io/en/latest/adv_traffic_manager/).

glopezdiest, 2024-06-21:
@GoodarzMehr @ccconquer No, the TrafficManager is created automatically once the first vehicle is registered to it, unless you actually call the `tm = client.get_trafficmanager(port)` first. You can just create a vehicle and use the set_autopilot(), and it won't cause any issues. Also, it doesn't matter if the simulation is synchronous or not. An asycnhronous TM will have a worse behavior, but it will not crash.

From what I understand, you tried doing the `set_autopilot()` without any port first, which used the default 8000 and that crashed? And what are you doing in the script apart from running the TM?

closed 2024-10-18 by glopezdiest