carla-simulator/carla #7826: Cloning into 'UnrealEngine'... remote: Repository not found. fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

omegaxy1124, 2024-06-19:
Dear people:
OS: Windows 11
can not run this thing:   git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git 
issue: Cloning into 'UnrealEngine'...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found
I am connected to my Git from Unreal Engine as it shows in the apps and accounts and I followed the instruction to a T and I have the organization badge on my profile but i just can't get it working it keeps giving me the same error: C:\>    git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git
Cloning into 'UnrealEngine'...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found
any help would be appreciated

JingyanXing, 2024-06-20:
the same for me

omegaxy1124, 2024-06-20:
Yeah hopefully they fix this soon cause i spent about 3 hours wracking my brain thinking i messed something up before i figured out this was outside my control.

PatrickPromitzer, 2024-06-20:
For me, it only works with a personal access token, but the last time I tried to clone CarlaUnreal was last week.
[https://github.com/settings/tokens](https://github.com/settings/tokens)

omegaxy1124, 2024-06-25:
Yep still broken, no clue if there is an alternative way to do this.

Fitz-Lee, 2024-06-28:
Hey Brothers. the Unreal Engine author replied!
[image]

rmrtcla, 2024-06-28:
Works for me now. I didn't get the invite to join the original EpicGames org on Github automatically (or maybe I was just too impatient), so I tried unlinking and then linking again my Epic account back to Github, and then it worked (about 3h ago). Thanks all.

omegaxy1124, 2024-06-29:
thank you everyone the problem is now fixed.

closed 2024-06-29 by omegaxy1124