carla-simulator/carla #7827: Can CARLA 0.9.15 （unbuntu remote headless server）use a map modified by CARLA 0.9.14（windows + UE4）？

ccconquer, 2024-06-20:
Thanks!!

CARLA version: 0.9.14
Platform/OS: windows

Problem you have experienced:
Can buildings be placed at specified locations via the Python API like actors, or can only be placed by modifying the map? Can CARLA 0.9.15 （unbuntu remote headless server）use a map modified by CARLA 0.9.14（windows + UE4），How should the modified map be uploaded to the remote headless server?

PatrickPromitzer, 2024-06-20:
Hi,
theoretical, if you have an Carla 0,9,14 map, you could copy the map files in the Carla 0.9.15 folder.
Maybe some object files were moved and can't be loaded (or cash the server), but it should work.

```
copy
carla_package_0.9.14/CarlaUE4/Content/Carla/Maps/YourMap -> carla_package_0.9.15/CarlaUE4/Content/Carla/Maps/YourMap
```

You could install the Carla Source version and add some buildings as actors. I have no idea how they would behavior if you drive into them or if they even stay on the map.
[https://carla.readthedocs.io/en/latest/tuto_content_authoring_vehicles/#creating-the-blueprint](https://carla.readthedocs.io/en/latest/tuto_content_authoring_vehicles/#creating-the-blueprint)
```
Now navigate to Content > Carla > Blueprints > Vehicles > VehicleFactory and double click this to open the Vehicle Factory.

Select the Vehicles node and expand the Vehicles item in the Default value section on the right hand side.

....

Press the plus icon to add your new vehicle. Scroll down to the last entry and expand it, it should be empty. 
Name the make and model of your vehicle and under the class section find your blueprint class that you created in the previous section. 
Leave the number of wheels as 4 and put the generation as 2. Compile and save. 
Do a global save for safety and you are now..ready to run your vehicle in a simulation.

```

ccconquer, 2024-07-02:
Thanks a lot for your explanation, it was very helpful to me!

closed 2024-11-06 by Blyron