carla-simulator/carla #7828: What is the path of a  new-added obstacle in blueprint_library?

huangchunan, 2024-06-20:
CARLA version:CARLA 0.9.15
Platform/OS:ubuntu 20.04
Questions ：
 I followed https://carla.readthedocs.io/en/0.9.15/tuto_A_add_props/ to add a new obstacle, but I cannot visit it in blueprint_library. It shows IndexError: the path is not found.
Any answer is welcome！
`obstacle_bp = blueprint_library.find('/Game/package01/Static/Static/shuipinga/shuipinga.shuipinga')`
[image]

PatrickPromitzer, 2024-06-21:
Hi,
The blueprint's name should be something similar to "static.prop.shuipinga"
startic.prop. + name

You can look at the list in Content/Carla/Blueprints/Props/propFactory

Madecu, 2024-06-21:
Hi @huangchunan,
it is exactly what Patrick said. It should be something like static.prop.shuipinga

Blyron, 2024-06-27:
It won't work. The prop factory does not support blueprints, only support static meshes itself so you cannot use your Blueprints for now. In next version of CARLA UE4 We are planning to have a blueprint factory but for now, prop factory is only able to spawn static meshes. Point the json file to the static mesh asset and not blueprint asset

huangchunan, 2024-07-02:
> It won't work. The prop factory does not support blueprints, only support static meshes itself so you cannot use your Blueprints for now. In next version of CARLA UE4 We are planning to have a [...]

Thanks! Do u know how to use the static mesh which is in the directory of UE4 ? It is not in the blueprints library.
[image]

huangchunan, 2024-07-05:
> If you want to use an Object with the Calrla API, but it isn't in the propFactory, you have to add the Object to the propFactory. You need to search for the "+" to add a new item, and then fill out the entries. One of the Entries is the path to the static mesh. You could be able to click on the entry and in the popup you can use the search to find the static mesh object.

I made a 3D model of watertube in Blender. I add it in the PropFactory. I can read it in the Blueprints library. But when it runs, I cannot see it. I set it in front of the car. Anything wrong with my model (fbx file)? When I drag the static mesh to the map. I can see the watertube.
[image]
[image]

Blyron, 2024-07-05:
I guess you are not spawnning it properly.

PatrickPromitzer, 2024-07-05:
Did you push the compile and the save button?
[image]

If you did, can you post the code where you spawn it?

huangchunan, 2024-07-08:
> Did you push the compile and the save button? [...]

I have compiled and saved it. The following is my code. The spawn_point is the ego car's location. I set the obstacle in front of the ego car. When I switch to a custom obstacle in blueprints library, it shows. However when I switch to my 3D model，it does not show. Any other things I missed?
`obstacle_bp = blueprint_library.find('static.prop.watertube1')` 
`blueprint_obs = obstacle_bp`       
`spawn_point.location.x -= 5`        
`spawn_point.location.z += 1`        
`self.obstacle = self.world.try_spawn_actor(blueprint_obs, spawn_point)`       
[image]