carla-simulator/carla #7830: Responding error from function get_wheel_steer_angle: Actor could not be found in the registry.  Actor Id: <xyz>

stevenbrills, 2024-06-20:
CARLA version: 0.9.13

Platform/OS: Ubuntu 20.04

Problem you have experienced: When running a simulation interfaced with a ROS mediated autonomy stack, the CARLA interface breaks when attempting to retrieve wheel steer angles of the car using the `get_wheel_steer_angle` method on the ego vehicle object. There is a line just before the call to this method that checks the list of actors returned by the CARLA world for the ego agent identified by the role name. The agent exists during this check. In the line that immediately follows, there is a call to the ego agent's "get_wheel_steer_angle" method with the location specified by "carla.VehicleWheelLocation.FR_Wheel" which fails with the response: Responding error from function get_wheel_steer_angle: Actor could not be found in the registry.  Actor Id: <xyz>. It is to be noted that this happens for some example scenarios in the scenario runner, (particularly the Opposite Direction) but not others so it does work some of the time. Even in the scenarios where it fails, the interface works as the car begins to move but then terminates after a few ticks of applying controls and getting the vehicle moving.

What you expected to happen: Since the ego agent is verified to exist in the world, I expected to be able to probe the agent for its wheel angles.

Steps to reproduce: Run the "ManeuverOppositeDirection_2" scenario with a Lincoln MKZ 2020. In the bridge interface that you use, try to call the "get_wheel_steer_angle" method on the ego agent during each tick.

Other information (documentation you consulted, workarounds you tried): Referred to other issues like this one #4553 and #7726 but it was not of much help.

generalchan825, 2024-09-23:
hi did you solve the issue?

sshawn9, 2025-01-02:
I wrote carla c++ client based on libcarla for my own simulation. Please check [this](https://github.com/carla-simulator/carla/issues/1939#issuecomment-2567488554), hope helps.