carla-simulator/carla #7833: Draw 3D bounding boxes for objects and actors

ccconquer, 2024-06-21:
Thanks !!

CARLA version: 0.9.14
Platform/OS: windows
Problem you have experienced: When I attempt to draw 3D bounding boxes for objects and actors, some bounding boxes are correct while others are not. For example, the bounding box representation for 'vehicle.tesla.cybertruck' is incorrect, and particularly tall buildings also show incorrect bounding boxes.
Code:
import glob
import sys
import os
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
import carla

from spawn_vehicle import SpawnVehicle
import numpy as np

def get_object_bbox_vertices(bbox):
    # Extract location, extent and rotation
    location = bbox.location
    extent = bbox.extent
    rotation = bbox.rotation
    # Convert rotation to radians
    roll = np.deg2rad(rotation.roll)
    pitch = np.deg2rad(rotation.pitch)
    yaw = np.deg2rad(rotation.yaw)
    # Create rotation matrix
    R = np.array([
        [np.cos(yaw) * np.cos(pitch), np.cos(yaw) * np.sin(pitch) * np.sin(roll) - np.sin(yaw) * np.cos(roll), np.cos(yaw) * np.sin(pitch) * np.cos(roll) + np.sin(yaw) * np.sin(roll)],
        [np.sin(yaw) * np.cos(pitch), np.sin(yaw) * np.sin(pitch) * np.sin(roll) + np.cos(yaw) * np.cos(roll), np.sin(yaw) * np.sin(pitch) * np.cos(roll) - np.cos(yaw) * np.sin(roll)],
        [-np.sin(pitch), np.cos(pitch) * np.sin(roll), np.cos(pitch) * np.cos(roll)]
    ])
    # Define the 8 vertices of the bounding box
    x = extent.x
    y = extent.y
    z = extent.z
    vertices = np.array([
        [x, y, z],
        [-x, y, z],
        [x, -y, z],
        [-x, -y, z],
        [x, y, -z],
        [-x, y, -z],
        [x, -y, -z],
        [-x, -y, -z]
    ])  
    # Rotate and translate vertices
    rotated_vertices = np.dot(vertices, R.T)
    translated_vertices = rotated_vertices + np.array([location.x, location.y, location.z])
    # Convert vertices to carla.Location objects
    carla_vertices = [carla.Location(x=vertex[0], y=vertex[1], z=vertex[2]) for vertex in translated_vertices]
    return carla_vertices

def draw_object_bbox(world, bbox):
    vertices = get_object_bbox_vertices(bbox)
    color = carla.Color(255, 0, 0)
    # Define lines to connect vertices
    lines = [
        (0, 1), (1, 3), (3, 2), (2, 0),  # Top face
        (4, 5), (5, 7), (7, 6), (6, 4),  # Bottom face
        (0, 4), (1, 5), (2, 6), (3, 7)   # Vertical lines
    ]
    # Draw lines
    for start, end in lines:
        world.debug.draw_line(vertices[start], vertices[end], thickness=0.1, color=color, life_time=1000)

def get_actor_bbox_vertices(actor):
    # Get actor's transform (location and rotation)
    transform = actor.get_transform()
    location = transform.location
    rotation = transform.rotation
    # Get actor's bounding box extent
    bbox = actor.bounding_box
    extent = bbox.extent
    # Convert rotation to radians
    roll = np.deg2rad(rotation.roll)
    pitch = np.deg2rad(rotation.pitch)
    yaw = np.deg2rad(rotation.yaw)
    # Define the 8 vertices of the bounding box (assuming extent is in x, y, z)
    x = extent.x / 2.0
    y = extent.y / 2.0
    z = extent.z / 2.0
    vertices = np.array([
        [x, y, z],
        [-x, y, z],
        [x, -y, z],
        [-x, -y, z],
        [x, y, -z],
        [-x, y, -z],
        [x, -y, -z],
        [-x, -y, -z]
    ])
    # Create rotation matrix
    R = np.array([
        [np.cos(yaw) * np.cos(pitch), np.cos(yaw) * np.sin(pitch) * np.sin(roll) - np.sin(yaw) * np.cos(roll), np.cos(yaw) * np.sin(pitch) * np.cos(roll) + np.sin(yaw) * np.sin(roll)],
        [np.sin(yaw) * np.cos(pitch), np.sin(yaw) * np.sin(pitch) * np.sin(roll) + np.cos(yaw) * np.cos(roll), np.sin(yaw) * np.sin(pitch) * np.cos(roll) - np.cos(yaw) * np.sin(roll)],
        [-np.sin(pitch), np.cos(pitch) * np.sin(roll), np.cos(pitch) * np.cos(roll)]
    ])
    # Rotate and translate vertices
    rotated_vertices = np.dot(vertices, R.T)
    translated_vertices = rotated_vertices + np.array([location.x, location.y, location.z])
    # Convert vertices to carla.Location objects
    carla_vertices = [carla.Location(x=vertex[0], y=vertex[1], z=vertex[2]) for vertex in translated_vertices]
    return carla_vertices

def draw_actor_bbox(world, actor):
    vertices = get_actor_bbox_vertices(actor)
    color = carla.Color(255, 0, 0)
    # Define lines to connect vertices
    lines = [
        (0, 1), (1, 3), (3, 2), (2, 0),  # Top face
        (4, 5), (5, 7), (7, 6), (6, 4),  # Bottom face
        (0, 4), (1, 5), (2, 6), (3, 7)   # Vertical lines
    ]
    # Draw lines
    for start, end in lines:
        world.debug.draw_line(vertices[start], vertices[end], thickness=0.1, color=color, life_time=1000)

def get_3d_bbox_height(bbox):
    height = 2 * bbox.extent.z  # Multiply by 2 to get the full height
    return height

other_vehicle_list = []
client = carla.Client('127.0.0.1', 2000)
client.set_timeout(2.0)
world = client.get_world()
spawn = SpawnVehicle(client,world)
ego_vehicle = spawn.ego_vehicle_create()
other_vehicle_list = spawn.other_vehicle_create()
object_bbox_set = []
forward_nearby_bboxes = []
object_bbox_set = world.get_level_bbs(carla.CityObjectLabel.Buildings)
object_bbox_set.extend(world.get_level_bbs(carla.CityObjectLabel.Fences))
object_bbox_set.extend(world.get_level_bbs(carla.CityObjectLabel.Static))

for bbox in object_bbox_set:
    if bbox.location.distance(ego_vehicle.get_transform().location) < 50:
        if get_3d_bbox_height(bbox) > 2:
            forward_vec = ego_vehicle.get_transform().get_forward_vector()
            ray = bbox.location - ego_vehicle.get_transform().location
            if forward_vec.dot(ray) >= 0:
                forward_nearby_bboxes.append(bbox)
                draw_object_bbox(world, bbox)

for actor in other_vehicle_list:
    bbox = actor.bounding_box
    dist = actor.get_transform().location.distance(ego_vehicle.get_transform().location)
    if dist < 50:
        if get_3d_bbox_height(bbox) > 2:
            forward_vec = ego_vehicle.get_transform().get_forward_vector()
            ray = ego_vehicle.get_transform().location - ego_vehicle.get_transform().location
            if forward_vec.dot(ray) >= 0:
                forward_nearby_bboxes.append(bbox)
                print('nearby_boxes', forward_nearby_bboxes)
                draw_actor_bbox(world,actor)

Result picture:
[image]
[image]

gsmario, 2024-06-21:
Hello.
These buildings are made with modules, it seems that it is only showing you some of those modules. As you can see in the image, all the pieces have their bounding box, so I have to assume that some part of your code is not working properly. 

[image]

The cybertruck vehicle (and many other) are also build like this.

PatrickPromitzer, 2024-06-21:
Hi,
for the cybertruck, it looks like you didn't multiply by 2
```
The value in each axis equals half the size of the box for that axis. extent.x * 2 would return the size of the box in the X-axis.
```
https://carla.readthedocs.io/en/latest/python_api/#carla.BoundingBox

The actor bounding boxes are not always accurate.
I wrote about it in #7007

gsmario, 2024-06-28:
I have been testing the error and I cannot reproduce it, the lights look fine to me. The source of the emissive textures is the same as version 9.14, so it shouldn't be a problem.
It may be that in the next release your problem will disappear.
Thanks for your time.

ccconquer, 2024-07-02:
> Hello. These buildings are made with modules, it seems that it is only showing you some of those modules. As you can see in the image, all the pieces have their bounding box, so I have to assume that some part of your code is not working properly.
> 
> ![image](https://private-user-images.githubusercontent.com/44405709/341738435-9003c7bc-6356-419f-b587-fc96e0d178b3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MjQ4NDgsIm5iZiI6MTcxOTkyNDU0OCwicGF0aCI6Ii80NDQwNTcwOS8zNDE3Mzg0MzUtOTAwM2M3YmMtNjM1Ni00MTlmLWI1ODctZmM5NmUwZDE3OGIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAyVDEyNDkwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI5Nzk4MDVhNDZlMDNjYjYzMjdjNmJkOWE4NTIzNWQyNTNhODdlY2ZmMzUxMmIzOWQ3OTZjODk1ZDA1ZWYzYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Wrl9PbNGha0c3_U2dd7BCGh6IW0E-ZOcXmkcbMIRu0g)
> 
> ![image](https://private-user-images.githubusercontent.com/44405709/341738213-e64fc214-0766-4622-a0e3-700df5f9df0e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MjQ4NDgsIm5iZiI6MTcxOTkyNDU0OCwicGF0aCI6Ii80NDQwNTcwOS8zNDE3MzgyMTMtZTY0ZmMyMTQtMDc2Ni00NjIyLWEwZTMtNzAwZGY1ZjlkZjBlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAyVDEyNDkwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxOTdkOWViYTlkZjRhMDMxMDgxNzZjMzczYzYzN2Q0OTEyY2JjZGYzMGZiYWZlNGM4ODQ4ZDhjYzBiZjU5YjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2Dfs2PmRL570DOFH8jDwX5v7VAzmy8wlQn08CRkg02Y)
> 
> The cybertruck vehicle (and many other) are also build like this, but it doesn't look like your tool is drawing the bounding boxes correctly.
> 
> Let me know if I can help you anyway.

Thank you so much! I think I will modify the code to correctly draw the bounding boxes for all modules.

gsmario, 2024-07-05:
Cool, thank you for your time and help!
I'm closing this issue.

closed 2024-07-05 by gsmario